raulmur/ORB_SLAM2 #114: orbslam2 raising static_assert when compiling with clang

OtacilioNeto, 2016-06-14:
Dears

When I try compile orbslam2 under FreeBSD 11 I'm getting a "static_assert failed". Searching the Internet I have found that this maybe can be a bug on orbslam that gcc don't catches (https://llvm.org/bugs/show_bug.cgi?id=15576).

Bellow is the error message: 

In file included from /usr/home/ota/workspace/orbslam2/src/LoopClosing.cc:21:
In file included from /usr/home/ota/workspace/orbslam2/include/LoopClosing.h:24:
In file included from /usr/home/ota/workspace/orbslam2/include/KeyFrame.h:24:
In file included from /usr/local/include/DBoW2/BowVector.h:14:
/usr/include/c++/v1/map:837:5: error: static_assert failed "Allocator::value_type must be same type as value_type"
    static_assert((is_same<typename allocator_type::value_type, value_type>::value),
    ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/home/ota/workspace/orbslam2/src/LoopClosing.cc:436:21: note: in instantiation of template class 'std::__1::map<ORB_SLAM2::KeyFrame
      *, g2o::Sim3, std::__1::less<ORB_SLAM2::KeyFrame *>, Eigen::aligned_allocator<std::__1::pair<const ORB_SLAM2::KeyFrame *,
      g2o::Sim3> > >' requested here
    KeyFrameAndPose CorrectedSim3, NonCorrectedSim3;
                    ^
/usr/home/ota/workspace/orbslam2/src/LoopClosing.cc:466:29: error: no viable overloaded operator[] for type 'KeyFrameAndPose' (aka
      'map<ORB_SLAM2::KeyFrame *, g2o::Sim3, std::less<KeyFrame *>, Eigen::aligned_allocator<std::pair<const KeyFrame *, g2o::Sim3> >
      >')
            NonCorrectedSim3[pKFi]=g2oSiw;
            ~~~~~~~~~~~~~~~~^~~~~
/usr/home/ota/workspace/orbslam2/src/LoopClosing.cc:470:39: error: use of undeclared identifier 'mit'; did you mean 'min'?
        for(KeyFrameAndPose::iterator mit=CorrectedSim3.begin(), mend=CorrectedSim3.end(); mit!=mend; mit++)
                                      ^~~
                                      min

OtacilioNeto, 2016-06-15:
The compiler of FreeBSD 11 is Clang.

cvcore, 2016-11-19:
I also bumped into static_assert error with clang 3.8 on ubuntu 16.04. The error I got is from Eigen:

> /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:745:3:
> error: 
>       static_assert failed
>       "YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY"
>   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);