raulmur/ORB_SLAM2 #117: SegFault on loop closure

finostro, 2016-06-16:
Hello, 

I am getting a segmentation fault when running Examples/Stereo/stereo_kitti with the sequence 00 of the kitti dataset. The error seems to come from the loopclosung procedure, specifically LoolClosing.cc at line 437: " CorrectedSim3[mpCurrentKF]=mg2oScw".

Here is the output of gdb with the error:

```
Thread 4 "stereo_kitti" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffcaf33700 (LWP 3543)]
ORB_SLAM2::LoopClosing::CorrectLoop (this=this@entry=0x7cba30) at /home/finostro/Code/ORB_SLAM2/src/LoopClosing.cc:437
437	    CorrectedSim3[mpCurrentKF]=mg2oScw;
(gdb) bt
#0  ORB_SLAM2::LoopClosing::CorrectLoop (this=this@entry=0x7cba30) at /home/finostro/Code/ORB_SLAM2/src/LoopClosing.cc:437
#1  0x00007ffff7b159ca in ORB_SLAM2::LoopClosing::Run (this=0x7cba30) at /home/finostro/Code/ORB_SLAM2/src/LoopClosing.cc:75
#2  0x00007ffff6e40c70 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff51bc6fa in start_thread (arg=0x7fffcaf33700) at pthread_create.c:333
#4  0x00007ffff68afb5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
(gdb) print mpCurrentKF
$1 = (ORB_SLAM2::KeyFrame *) 0x1f641ab0
(gdb) print CorrectedSim3
$2 = std::map with 1 elements = {[0x1f641ab0] = {
    r = {<Eigen::QuaternionBase<Eigen::Quaternion<double, 0> >> = {<Eigen::RotationBase<Eigen::Quaternion<double, 0>, 3>> = {<No data fields>}, <No data fields>}, 
      m_coeffs = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<Eigen::internal::special_scalar_op_base<Eigen::Matrix<double, 4, 1, 0, 4, 1>, double, double, Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 3>, false>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {0, 0, 0, 1}}}}, <No data fields>}}, 
    t = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::internal::special_scalar_op_base<Eigen::Matrix<double, 3, 1, 0, 3, 1>, double, double, Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>, false>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {0, 0, 0}}}}, <No data fields>}, s = 1}}
(gdb) print mg2oScw 
$3 = {
  r = {<Eigen::QuaternionBase<Eigen::Quaternion<double, 0> >> = {<Eigen::RotationBase<Eigen::Quaternion<double, 0>, 3>> = {<No data fields>}, <No data fields>}, 
    m_coeffs = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<Eigen::internal::special_scalar_op_base<Eigen::Matrix<double, 4, 1, 0, 4, 1>, double, double, Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 3>, false>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {-0,011991941191092759, -0,71804767883081744, 0,005538132639597816, 
              0,69586857093937493}}}}, <No data fields>}}, 
  t = {<Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::MatrixBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::DenseBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<Eigen::internal::special_scalar_op_base<Eigen::Matrix<double, 3, 1, 0, 3, 1>, double, double, Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>, false>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 3>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 1>> = {<Eigen::DenseCoeffsBase<Eigen::Matrix<double, 3, 1, 0, 3, 1>, 0>> = {<Eigen::EigenBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, m_storage = {m_data = {array = {90,685019375400898, 0,11166814559853692, -1,3507948197562571}}}}, <No data fields>}, s = 1}
(gdb) 
```

Any ideas on how to fix this?

Thanks!!

finostro, 2016-06-16:
I reran gdb building on debug instead of RelWithDebInfo and it gave me an eigen alignment assertion fail, so I fixed it by moving the  EIGEN_MAKE_ALIGNED_OPERATOR_NEW in the Thirdparty/g2o/g2o/types/sim3.h file as follows: 
From:
´´´
  struct Sim3
  {
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW

  protected:
    Quaterniond r;
    Vector3d t;
    double s;

public:
    
    Sim3()
    {
      r.setIdentity();
      t.fill(0.);
      s=1.;
    }

    Sim3(const Quaterniond & r, const Vector3d & t, double s)
      : r(r),t(t),s(s)
    {
    }

´´´
to:
´´´
  struct Sim3
  {
    

  protected:
    Quaterniond r;
    Vector3d t;
    double s;

public:
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
    Sim3()
    {
      r.setIdentity();
      t.fill(0.);
      s=1.;
    }

    Sim3(const Quaterniond & r, const Vector3d & t, double s)
      : r(r),t(t),s(s)
    {
    }

´´´

jpiat, 2016-11-23:
Hi,

i encountered the same issue but the proposed solution does not seem to work for me. Any idea of how to investigate ?
Thanks

wowitsmrinal, 2016-12-02:
Was this ever resolved? @jpiat @finostro @raulmur

chaiein, 2017-01-13:
Above solution doesnt work, Even I faced the same issue.

jpiat, 2017-01-13:
I got it to work by removing -march=native in CMakeList.text and compiling
in RelDebInfo instead of Release. The bug seems to be caused by alignment
issue and will appear depending on your architecture.

chaiein, 2017-01-13:
In CMakeLists.txt  of ORB_SLAM2 tried following options
1.Removed -march=native in 
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")

3. SET(CMAKE_BUILD_TYPE RelWithDebInfo) - not solved
4. SET(CMAKE_BUILD_TYPE RelDebInfo) - not solved

jpiat, 2017-01-13:
This solve the issue but could lead the compiler to not fully use your specific processor. I'am still investigating a solution to this problem but this seems to be a g2o problem not something in ORB_SLAM2

jpiat, 2017-01-13:
If you install lastest stable release of Eigen, ORB_SLAM2 will not even start since Eigen will detected (assert) the alignement issue. The bus is clearly related with Sim3 type in g2o, but i have no clue of how to fix it. Surprisingly this problem does not seem to affect everybody, it might be interesting to know what the difference in setup for a working and non-working implementation.

I'am using GCC 5.4.0, Eigen 3.3, OpenCV 3.1

raulmur, 2017-01-13:
Check the latest commits

puiglm, 2017-09-15:
@jpiat Did you find a solution to the problem? I found that Eigen 3.3 was assuming that on 64-bit systems, glibc's malloc returns 16-byte-aligned pointers. You can see the Memory.h for more detail. I forced Eigen to do its own alignment and that allows the program at least start. Then I have a problem with SEQuat.

UannaFF, 2019-03-13:
I downloaded and ran just fine with opencv3.2 but when i changed to opencv3.1 I started getting this problem. I think this is the problem, don't know why but that's the only thing I changed.