raulmur/ORB_SLAM2 #118: Warning when building nodes for ROS examples

ap108, 2016-06-19:
Hi 
When I run 'make -j' to build nodes for ROS examples.
I get

. . . 
[ 33%] [100%] [100%] Building CXX object CMakeFiles/Stereo.dir/src/ros_stereo.cc.o 
Building CXX object CMakeFiles/Mono.dir/src/ros_mono.cc.o 
Building CXX object CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o 
Linking CXX executable ../Mono 
/usr/bin/ld: warning: libopencv_core.so.3.0, needed by ../../../../lib/libORB_SLAM2.so, may conflict with libopencv_core.so.2.4 
[100%] Built target Mono 
Linking CXX executable ../RGBD 
Linking CXX executable ../Stereo 
/usr/bin/ld: warning: libopencv_core.so.3.0, needed by ../../../../lib/libORB_SLAM2.so, may conflict with libopencv_core.so.2.4 
[100%] Built target RGBD 
/usr/bin/ld: warning: libopencv_imgproc.so.3.0, needed by ../../../../lib/libORB_SLAM2.so, may conflict with libopencv_imgproc.so.2.4 
[100%] Built target Stereo 

If  I try to run the example with rosrun I get:

. . . 

OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp, line 740
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp:740: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName

Aborted (core dumped)
 
I guess I am not linking opencv correctly here but I don't how to do this.

I am using ROS Indigo, Ubuntu 14.04.

Thank you

I0x0I, 2016-08-26:
See #103

luxinglong, 2017-05-04:
@ap108 I also meet this problem. I use ROS indigo+Ubuntu 14.04+opencv 3.1.0. Did you figure it out? need help.

ap108, 2017-05-05:
I think I tried
FIND_PACKAGE(OpenCV 2.4.8 EXACT REQUIRED)
in CMakeLists.txt. 
In your case it would be 3.1.0 instead of 2.4.8. Hope it helps.

Harry-675, 2017-06-03:
I met these question too.But I can't solve this problem by change the CMakeList.txt.
Is there any other solutions?

dominikj93, 2017-06-23:
@Harry-675 Hey! I have the same problem, changing the CMakeList.txt also does not work form me. Have you found the solution?

Harry-675, 2017-06-27:
@dominikj93 No.But I found another way to use camera stream.You don't have to use the ROS.Just change the mono_tum.cpp,and use opencv to get camera stream and translate it into frame as a dataset.

lanyouzibetty, 2017-07-18:
@Harry-675 Hello! Have you solved the problem? I have tried change the CMakeList.txt, to make it use 3.1.0 opencv，but it did not work.

Harry-675, 2017-07-18:
@lanyouzibetty I've solved it by editing the mono_tum.cpp.So I don't use the ROS to run a camera example.Just changed the mono_tum by using opencv video capture to get camera stream.

lanyouzibetty, 2017-07-19:
@Harry-675 Thank you for your reply. So now you can run the orb_slam real time without ROS?

lanyouzibetty, 2017-07-19:
@Harry-675 Thank you for your advice, I have modified it successfully.