raulmur/ORB_SLAM2 #121: How to get the scale value of TUM evaluation?

skylook, 2016-06-27:
Hi, I run the mono_tum code to get the TUM test results, but in their website, there should be a scale for evaluation. 
How can I get the scale value? Thanks.

Beewe, 2016-06-29:
Have a look here:
https://github.com/raulmur/ORB_SLAM/issues/28#issuecomment-116040935

MichaelGrupp, 2017-10-27:
If you're able to register (associate) your trajectories, you can do a scale alignment to the ground truth in closed form. One method suited for this is [Umeyama's method](http://ieeexplore.ieee.org/abstract/document/88573/) - this gives you a similarity transformation (Sim(n)) that aligns two n-dimensional point sets with respect to translation, rotation and scale.

You can checkout my SLAM evaluation package where I integrated the option to use these kinds of alignment (`--align --correct_scale` or short: `-as` for Sim(3) alignment). In your case, you would run something like:
```
evo_ape tum <groundtruth.txt> <your_trajectory.txt> -as --plot
```

MichaelGrupp, 2017-10-27:
If you're able to register (associate) your trajectories, you can do a scale alignment to the ground truth in closed form. One method suited for this is [Umeyama's method](http://ieeexplore.ieee.org/abstract/document/88573/) - this gives you a similarity transformation (Sim(n)) that aligns two n-dimensional point sets with respect to translation, rotation and scale.

You can check out my SLAM/odometry evaluation package [evo](https://github.com/MichaelGrupp/evo) where I integrated the option to use these kinds of alignment (`--align --correct_scale` or short: `-as` for Sim(3) alignment). In your case, you would run something like:
```
evo_ape tum <groundtruth.txt> <your_trajectory.txt> -as --plot
```
for the absolute error or 
```
evo_rpe tum <groundtruth.txt> <your_trajectory.txt> -as --plot
```
for the relative error.