raulmur/ORB_SLAM2 #123: crash while track lost soon after initialisation and reset

flankechen, 2016-06-30:
I find the program would crash if I shake my camera or put my hand right in front just after initialize.The programe goes into 
```
        // Reset if the camera get lost soon after initialization
        if(mState==LOST)
        {
            if(mpMap->KeyFramesInMap()<=5)
            {
                cout << "Track lost soon after initialisation, reseting..." << endl;
                mpSystem->Reset();
                return;
            }
        }
```

I find that in this situation, in ` void Tracking::UpdateLastFrame()` the size of `mlRelativeFramePoses` would be just equal to one, which is not sufficient and sufficient.

I am not quite sure why this is happening but I fixed this by checking whether this size is less than 2.
```
	if (mlRelativeFramePoses.size() < 2)
	{
		cout << "mlRelativeFramePoses size less than 2" << endl;
		return;
	}
```

OrlandoLara22, 2019-03-18:
Thanks @flankechen I was struggling with this problem too. I couldn't find any solution to this problem online except for this post. I tried to fix the problem myself and manged to reduce the problem to this function but couldn't quite pinpoint what was making the program crash. I noticed that this problem doesn't occur once the system has more than 5 keyframes and that isn't so hard to achieve so I guess not many people were interested in solving this problem. I'm glad to know that I found at least one other person that tried to fix this. Thank you so much. You're solution worked for me

jyp0802, 2019-10-16:
Thanks! I had this problem as well and this fixed it :)

popsheng, 2019-11-14:
add code:
```
mnLastRelocFrameId = 0;
mVelocity = cv::Mat();    // this line can fix the bug.
```
to Tracking::Reset()

ns15417, 2020-01-09:
@popsheng did you solved the bug by doing this fix? I was also wondering why mVelocity has never been cleared cuz once the system is reset, it would start from TrackwithMotionModel();

laxnpander, 2020-06-03:
I was having an issue, that after manual reset the Tracking worked for a few frames but then failed again very quickly soon after. When start fresh from exactly this point, tracking was working perfectly fine. The recommended lines of @popsheng fixed this issue for me, thanks for that!

mcarletti, 2020-06-15:
Thank you @flankechen! Could you precise where I need to put your code? As far as I understood, I should add the `mlRelativeFramePoses` size check at the beginning of Tracking::UpdateLastFrame(). Thank you

hustchenli, 2020-08-05:
@flankechen Thanks for your solution. I added your code into the end of `void Tracking::UpdateLastFrame()`. But I still get the problem when the camera moves too fast and the program crashed. Here is the error:

```
Depth Threshold (Close/Far Points): 2.5
New map created with 678 points
OpenCV Error: Assertion failed (0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows) in Mat, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 483
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:483: error: (-215) 0 <= _rowRange.start && _rowRange.start <= _rowRange.end && _rowRange.end <= m.rows in function Mat

Aborted (core dumped)
```

hustchenli, 2020-08-05:
I just found the reason. Nothing to do with the program. I just created a publisher to output the camera_pose as rosmsg. And when track is lost, camera_pose will be an empty matrix, which could cause error.