raulmur/ORB_SLAM2 #124: I succeeded building but have some problem ablout Map viewer

zzzzzq, 2016-07-07:
Hello. I have built the project, but when I run it there are some problem about Map Viewer. The whole window is black and there is nothing in it.  But the Current Frame window works well. Can anyone tell me what is the problem? I guess it may be about Pangolin, but I have reinstalled several times and the problem is still there.

following is my snapshot of Map Viewer and Current Frame
[image]

zzzzzq, 2016-07-07:
The black window is my Map Viewer.

qianwenjun0801, 2016-08-17:
hi, what mean of your blue point? @zzzzzq

zzzzzq, 2016-11-22:
Sorry to response so late...... The blue point is because I use xtion and ORB_SLAM is in RGBD mode. If you use Mono there are only green points.

zzzzzq, 2016-11-22:
This problem is because I forgot to install Pangolin. When I successfully install Pangolin I can see the point cloud and keyframes.