raulmur/ORB_SLAM2 #130: Trade performance for Accuracy

drewsipher, 2016-07-15:
Hey there,
I'm super impressed by all the work you've done. The library is great. 

I did have a question though; Is there any way to improve accuracy, even if it's at the cost of performance. I am running pre-recorded samples, and I need the highest accuracy for localization possible. Looped samples work very well of course so we try to produce those as much as possible. The samples are 640x480. Would higher resolutions help? How much can I affect accuracy with the number of features or other parameters? Are there parameters for increasing the iterations or threshold size for local or global BA?

Thank you for any information you can share.

AlejandroSilvestri, 2017-02-02:
Hi,

Yes, higher resolution improve accuracy.

More keyframes help.  One way is messing the code to produce more keyframes, another is playing the video forward and backward many times on some spot in need, new keyframes will appear.

In Optimizer::LocalBundleAdjustment you can increase number of iterations and reduce threshold.  Optionally, you cam replace Local BA by Global BA, which will work slower as the map growth.

drewsipher, 2017-02-07:
Hey Alejandro,
Thanks for the suggestions. Unfortunately I'm not using Orb SLAM anymore so I'm going to close this issue.

closed 2017-02-07 by drewsipher

haithink, 2017-12-06:
@drewsipher  So now which system are you using ?