raulmur/ORB_SLAM2 #132: Fail to build ORB_SLAM2 with Eigen3 (repository of Ubuntu 16.04)

jiapei100, 2016-07-26:
Hi, I obtained the following error message:

>  ...../ORB_SLAM2/src/Optimizer.cc:1244:1:   required from here
> /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:745:3: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
>    EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
>    ^
> CMakeFiles/ORB_SLAM2.dir/build.make:350: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed

Any suggestions?

Thank you
Pei

romdos, 2016-07-29:
I had the same problem. It was partially solved. Look at #https://github.com/raulmur/ORB_SLAM/issues/119

jiapei100, 2016-07-31:
Yes, Thank you...
 
> SparseMatrix::Index -> int
is my solution...

Cheers
Pei

closed 2016-07-31 by jiapei100