raulmur/ORB_SLAM2 #133: A question about aligning the keyframe trajectories with ground truth

hwb0314, 2016-07-27:
@raulmur 
Dear Mur-Artal:

Recently I've read your paper "ORB-SLAM: A Versatile and Accurate Monocular SLAM System", in which you get impressive results compared with other state-of-art SLAM systems. I want to say thank you for your work.

However, here is a problem troubling me for a long time. In the section VIII. EXPERIMENTS, B. Localization Accuracy in the TUM RGB-D Benchmark, you mention that in order to compare trajectory with the ground truth, you align the keyframe trajectories using a similarity transformation, as scale is unknown, and measure the absolute trajectory error. 

I want to wander what exact similarity transformation you used, may I just change the scale of the trajectories to get a minimal absolute trajectory error? I use the "evaluate_ate.py" script, one of  which automated evaluation tool published by TUM ( "A Benchmark for the Evaluation of RGB-D SLAM Systems", web: http://vision.in.tum.de/data/datasets/rgbd-dataset/tools#evaluation ), to caculate the ATE. When I change the scale factor, I get different RMSE. 

For example, I use ORB-SLAM2 to get a trajectory of fr3_long_office. When setting the scale factor to 1.0, I get RMSE=122.37cm, however I get RMSE=1.6cm with scale=2.45, which better than your result(RMSE=3.45cm) of your paper. Is it right for me to change the scale factory to get a minimal ATE for every database sequence? Or should I fixed the scale factory of the sequences captured by the same camera? 

Thanks again for your reading my email. I do appreciate your generous sharing. And have a nice day! Looking forward to your reply.

[image]

[image]

marcusvgomez, 2016-09-13:
Did you get any response to this?

raulmur, 2016-09-14:
https://github.com/raulmur/evaluate_ate_scale

closed 2016-09-14 by raulmur