raulmur/ORB_SLAM2 #135: Core dumped Error ... Eigen 3.2.9???

jiapei100, 2016-07-31:
Hi, all:

My current configuration:
OS: Ubuntu 16.04
Compiler: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.1) 5.4.0 20160609
Eigen: 3.2.9 (built manually. Not the default eigen 3.3-beta1-e from repository )

> ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
> This program comes with ABSOLUTELY NO WARRANTY;
> This is free software, and you are welcome to redistribute it
> under certain conditions. See LICENSE.txt.
> 
> Input sensor was set to: Monocular
> 
> Loading ORB Vocabulary. This could take a while...
> Vocabulary loaded!
> 
> 
> Camera Parameters: 
> - fx: 517.306
> - fy: 516.469
> - cx: 318.643
> - cy: 255.314
> - k1: 0.262383
> - k2: -0.953104
> - k3: 1.16331
> - p1: -0.005358
> - p2: 0.002628
> - fps: 30
> - color order: RGB (ignored if grayscale)
> 
> ORB Extractor Parameters: 
> - Number of Features: 1000
> - Scale Levels: 8
> - Scale Factor: 1.2
> - Initial Fast Threshold: 20
> - Minimum Fast Threshold: 7
> 
> -------
> Start processing sequence ...
> Images in the sequence: 756
> 
> init done
> opengl support available
> New Map created with 95 points
> *** Error in `./Examples/Monocular/mono_tum': free(): invalid pointer: 0x000000000f7ef4a0 ***
> ======= Backtrace: =========
> /lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x7f5b68314725]
> /lib/x86_64-linux-gnu/libc.so.6(+0x7ff4a)[0x7f5b6831cf4a]
> /lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f5b68320abc]
> /usr/local/lib/libg2o.so(_ZN3g2o15VertexSE3ExpmapD0Ev+0x23)[0x7f5b67e2b7a3]
> /usr/local/lib/libg2o.so(_ZN3g2o10HyperGraph5clearEv+0x79)[0x7f5b67e49a79]
> /usr/local/lib/libg2o.so(_ZN3g2o16OptimizableGraphD1Ev+0x1e)[0x7f5b67e553ee]
> /home/jiapei/Downloads/computervision/ORB_SLAM2/lib/libORB_SLAM2.so(_ZN9ORB_SLAM29Optimizer16BundleAdjustmentERKSt6vectorIPNS_8KeyFrameESaIS3_EERKS1_IPNS_8MapPointESaIS9_EEiPbmb+0xcc8)[0x7f5b6a2353f8]
> /home/jiapei/Downloads/computervision/ORB_SLAM2/lib/libORB_SLAM2.so(_ZN9ORB_SLAM29Optimizer22GlobalBundleAdjustemntEPNS_3MapEiPbmb+0x67)[0x7f5b6a235667]
> /home/jiapei/Downloads/computervision/ORB_SLAM2/lib/libORB_SLAM2.so(_ZN9ORB_SLAM28Tracking25CreateInitialMapMonocularEv+0x495)[0x7f5b6a1d2b75]
> /home/jiapei/Downloads/computervision/ORB_SLAM2/lib/libORB_SLAM2.so(_ZN9ORB_SLAM28Tracking23MonocularInitializationEv+0x78d)[0x7f5b6a1d3dfd]
> /home/jiapei/Downloads/computervision/ORB_SLAM2/lib/libORB_SLAM2.so(_ZN9ORB_SLAM28Tracking5TrackEv+0x1437)[0x7f5b6a1d6d27]
> /home/jiapei/Downloads/computervision/ORB_SLAM2/lib/libORB_SLAM2.so(_ZN9ORB_SLAM28Tracking18GrabImageMonocularERKN2cv3MatERKd+0x1c8)[0x7f5b6a1d8fe8]
> /home/jiapei/Downloads/computervision/ORB_SLAM2/lib/libORB_SLAM2.so(_ZN9ORB_SLAM26System14TrackMonocularERKN2cv3MatERKd+0x106)[0x7f5b6a1c14f6]
> ./Examples/Monocular/mono_tum(main+0x839)[0x403249]
> /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f5b682bd830]
> ./Examples/Monocular/mono_tum(_start+0x29)[0x403d39]
> ======= Memory map: ========
> 00400000-00406000 r-xp 00000000 08:13 8400602                            /usr/local/share/orb_slam2/Examples/Monocular/mono_tum
> 00605000-00606000 r--p 00005000 08:13 8400602                            /usr/local/share/orb_slam2/Examples/Monocular/mono_tum
> 00606000-00607000 rw-p 00006000 08:13 8400602                            /usr/local/share/orb_slam2/Examples/Monocular/mono_tum
> 01d6a000-0fe1e000 rw-p 00000000 00:00 0                                  [heap]
> 40424000-40426000 r-xs 00000000 08:13 11286717                           /tmp/.glUHntTW (deleted)
> 40bd8000-40c78000 rw-p 00000000 00:00 0 
[... 220 lines cut ...]
> 7f5b20019000-7f5b20024000 r--s 00000000 08:13 4456542                    /var/cache/fontconfig/945677eb7aeaf62f1d50efc3fb3ec7d8-le64.cache-6
> 7f5b20024000-7f5b2002a000 r--s 00000000 08:13 4456524                    /var/cache/fontconfig/2cd17615ca594fa2959ae173292e504c-le64.cache-6
> 7f5b2002a000-7f5b2002b000 r--s 00000000 08:13 4456521                    /var/cache/fontconfig/0d8c3b2ac0904cb8a57a757ad11a4a08-le64.cache-6
> 7f5b2002b000-7f5b2002f000 r--s 00000000 08:13 4478347                    /var/cache/fontconfig/6d41288fd70b0be22e8c3a91e032eec0-le64.cache-6
> 7f5b2002f000-7f5b20033000 r--s 00000000 08:13 4476992                    /var/cache/fontconfig/de156ccd2eddbdc19d37a45b8b2aac9c-le64.cache-6
> 7f5b20033000-7f5b20048000 r--s 00000000 08:13 4456518                    /var/cache/fontconfig/04aabc0a78ac019cf9454389977116d2-le64.cache-6
> 7f5b20048000-7f5b20049000 r--s 00000000 08:13 4456523                    /var/cache/fontconfig/1ac9eb803944fde146138c791f5cc56a-le64.cache-6
> 7f5b20049000-7f5b2004a000 r--s 00000000 08:13 4469240                    /var/cache/fontconfig/b95bc8ffbebda2bbdae4265e45b8178d-le64.cache-6
> 7f5b2004a000-7f5b2004e000 r--s 00000000 08:13 4456528                    /var/cache/fontconfig/385c0604a188198f04d133e54aba7fe7-le64.cache-6
> 7f5b2004e000-7f5b2004f000 r--s 00000000 08:13 4469238                    /var/cache/fontconfig/9c956a7723ca69a44b382d9179c9802f-le64.cache-6
> 7f5b2004f000-7f5b20050000 r--s 00000000 08:13 4456555                    /var/cache/fontconfig/dc05db6664285cc2f12bf69c139ae4c3-le64.cache-6
> 7f5b20050000-7f5b20051000 r--s 00000000 08:13 4469236                    /var/cache/fontconfig/5d1cca7074f29429a8d18692746c2426-le64.cache-6
> 7f5b20051000-7f5b20054000 r--s 00000000 08:13 4456538                    /var/cache/fontconfig/767a8244fc0220cfb567a839d0392e0b-le64.cache-6
> 7f5b20054000-7f5b20055000 r--s 00000000 08:13 4477690                    /var/cache/fontconfig/4794a0821666d79190d59a36cb4f44b5-le64.cache-6
> 7f5b20055000-7f5b20056000 r--s 00000000 08:13 4469235                    /var/cache/fontconfig/9eae20f1ff8cc0a7d125749e875856bd-le64.cache-6
> 7f5b20056000-7f5b20059000 r--s 00000000 08:13 4469234                    /var/cache/fontconfig/bf2c1853a9e9b00bb02fe2e9bcf1e201-le64.cache-6
> 7f5b20059000-7f5b2005e000 r--s 00000000 08:13 4456541                    /var/cache/fontconfig/8801497958630a81b71ace7c5f9b32a8-le64.cache-6
> 7f5b2005e000-7f5b20062000 r--s 00000000 08:13 4476991                    /var/cache/fontconfig/c57959a16110560c8d0fcea73374aeeb-le64.cache-6
> 7f5b20062000-7f5b20063000 r--s 00000000 08:13 4456548                    /var/cache/fontconfig/bab58bb527bb656aaa9f116d68a48d89-le64.cache-6Aborted (core dumped)
> jiapei:orb_slam2$

Core dumped ...  Can anybody give me some help?

Cheers
Pei

jiapei100, 2016-10-20:
No idea how I figure this issue... But now, all problems gone....

Cheers

closed 2016-10-20 by jiapei100

shjtdxmhy, 2017-04-27:
How did you solve this problem?

AnthonyChuah, 2017-07-04:
I got this problem as well: anyone have any idea how this was solved?

JiankaiSun-SJTU-MVIG-training, 2017-08-23:
I also got this problem :(

cvDreamer, 2018-02-22:
I got the same problem, any solution?

mojovski, 2018-02-22:
Compile with debug run
gdb --args yourbinary parameters
then enter
run
In the gdb console.
When it crashes, enter
where
and you will see the source of the problem

melodybinbin, 2018-08-21:
https://github.com/raulmur/ORB_SLAM2/issues/547

melodybinbin, 2018-08-21:
https://github.com/raulmur/ORB_SLAM2/issues/547
You can try this. However, I have not solved it throuth the old version of Pangolin at https://github.com/stevenlovegrove/Pangolin/tree/8b8b7b96adcf58ac2755dedd3f681fc512385af0

shenlan1994, 2020-04-11:
Try to remove CMakeLists.txt "march=native" Optimization options，It may be help for you .

YangZhouDL, 2022-04-03:
Do you mean this record"QObject::~QObject: Timers cannot be stopped from another thread" does not have any influence since the trajectory has been saved?

nickxiang0306, 2022-07-15:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/452 maybe this issue could help you fix this segfault bug. And I met the bug when running orb-slam3.