raulmur/ORB_SLAM2 #139: What is Vocabulary/ORBvoc.txt for?

jiapei100, 2016-08-04:
Sorry for my entry-level question, but what is **Vocabulary/ORBvoc.txt** for?
Can I just ignore this parameter during testing my own camera?

Cheers

chwimmer, 2016-08-05:
Very basic question. But your answers is in here: [Bags of Binary Words for Fast Recognition in Image Sequences](http://webdiis.unizar.es/~dorian/papers/GalvezTRO12.pdf)

So no you cant ignore it.

jiapei100, 2016-08-05:
@chwimmer 
Thank you chwimmer, for your very prompt reply. So, Vocabularly/ORBvoc.txt cannot be ignored. But, when I use my own camera or captured images to start a fresh testing, should I use the same Vocabulary/ORBvoc.txt? Or, should I use my own Vocabulary/ORBvoc.txt?

Thanks...

chwimmer, 2016-08-05:
Just use the same Vocabulary because its taken from a huge sets of datasets and works pretty good.
I recommend to use [this pull request](https://github.com/raulmur/ORB_SLAM2/pull/21) because the Vocabulary is loaded faster but it also works with the ORBvoc.txt

jiapei100, 2016-08-05:
@chwimmer 
Ok, make sense.... Current Vocabulary/ORBvoc.txt is trained out from a huge database, which is more or less a kind of generic ... I wonder how I can train my own **Vocabulary/ORBvoc.txt**? Any tutorial? 

- How to collect sample data?
- How to train? which command?
- Anything else that we should pay attention to? etc.

Cheers

rikpires, 2016-08-09:
Yes you could train your own data set. Please refer to [https://github.com/dorian3d/DBoW2](url). 
Make sure to collect enough sample data.

qianwenjun0801, 2016-08-11:
@jiapei100  you should save lots of msg  as voc.txt/bin saved when you load the camera and run ORB_SLAM2, and then train these msg.., the program is not very difficult, just follow the algorithm which can been seen in   "DBoW2/TemplatedVocabulary.h "

Fatmoham, 2017-07-19:
In your paper "Fast Relocalisation and Loop Closing in Keyframe-Based SLAM" you created vocabulary using the data-set Bovisa 2008-09-01. This Vocabulary is taken from a huge sets of data and works pretty good for indoor and outdoor SLAM.
If I want to use the same program ORBSLAM2  in completely different environment rather than indoor and outdoor environment, are the ORBvoc.text still good to use? Or I need to generate my own Vocabulary for the data-sets I collected from that environment?

arifle, 2017-11-10:
@Fatmoham 
Have you already fixed it? 
I am asking myself the same question but have no answer and I come to search and find this. 
I don't exactly understand what the meaning of the 'large dataset' is. Does it mean that database is somehow common to most of the natural scene or it mean the database is a large one (or large enought) wrt all the experimental datasets used by ORBSLAM?

katey24, 2021-05-18:
Thank you for this forum.. it help me to test my samples in ORBSLAM research.

akshaygulabrao, 2022-07-29:
Thank you for this question.