raulmur/ORB_SLAM2 #140: How to change PATH_TO_SEQUENCE_FOLDER to a real-time Camera streaming data?

jiapei100, 2016-08-04:
Hi, sorry to bug you again... 

1) It looks like your data is really for a fusion data, namely:
vision data + accelerator data
and I guess this is why the EKF is adopted, right?
That's why under TUM, there are both rgb.txt (ignore depth.txt for now), as well as accelerometer.txt , right?
My first questions is: is your code possible to work for Vision only, even without the accelerator?

2) No matter it's able to for vision ONLY or both vision + accelerator
My second question is: how to make it work for the real thing (for example, my cell phone comes with accelerator as well as the vision), instead of just the downloaded data?
I'm imaging if it's as simple as change PATH_TO_SEQUENCE_FOLDER to another parameter will do?

Thanks
Pei

flankechen, 2016-09-13:
Hi Jia Pei, surprise to see you here!
I have been working with ORB_SLAM2 for a while. 
1. I couldn't find fusion things in this work. And one of the major thing I am facing is, syn with vision data and IMU. 

2. check stereo_video project example, ORB_SLAM2 have really nice architecture, you can add a new project in cmake and get image from opencv, and feed the SLAM system. for cell phones, it looks like the right timing to dive into, computing power is not sufficient. others achieve really low fps. 

Chen Zhizhao

jiapei100, 2016-10-02:
@flankechen 

Hi, Zhizhao... How is everything?
Thanks for your reply... ^_^

Pei

closed 2016-10-02 by jiapei100