raulmur/ORB_SLAM2 #143: Inconsistency and the dependency on number of features

HDaoud, 2016-08-12:
Hello Raul!

Very good work on ORB SLAM, and thank you for sharing your code!

I have a question regarding the reason behind getting different results every time I run the mono_tum although I'm using same configuration and same dataset, this happens specifically with **_rgbd_dataset_freiburg2_large_with_loop_** some time it initialize immediately and some times it takes until the camera reach the teddy bear, another problem is when lighting change some times it lost track, and sometimes it doesn't! also when reaching the wall,..etc. 

I tried to play with number of features and it seems to affect the results and performance alot! I appreciate explaining more on the reasons behind these issues.

All the best!
Hayyan

odiseo123, 2016-08-17:
Hi Raul and Hayyan,
I have experienced the same behavior on my own recorded videos, i.e. random initialization times on the same sequences and sensitivity to illumination changes coming from auto-exposure settings in the camera. 
Given that the overall tracking performance depends strongly on the initial map, it will be great if you could comment on how to make that stage particularly robust.

Thanks a lot for this amazing code. With kind regards,
Andres

AlejandroSilvestri, 2016-10-18:
Initialization uses Random Sample Consensus (Ransac) which is random, non deterministic, so its output varies from one time to another.

naja1987, 2017-05-19:
please also check #45