raulmur/ORB_SLAM2 #145: Build Issue Ubuntu 14.04 with Dependencies Installed

MCoder1, 2016-08-17:
Hi,

I think I have all my dependencies installed properly on a fresh version of Ubuntu. I have Eigen 3, OpenCV 2.4.11, installed LAPACK and BLAS, and Pangolin. I saw another issue with similar messages, but no responses/solutions... Any ideas?

> Configuring and building Thirdparty/DBoW2 ...
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found CUDA: /usr/local/cuda (found suitable exact version "7.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/roots/user1/ORB_SLAM2/Thirdparty/DBoW2/build
Scanning dependencies of target DBoW2
[ 16%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/BowVector.cpp.o
[ 33%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/FORB.cpp.o
[ 50%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/FeatureVector.cpp.o
[ 66%] Building CXX object CMakeFiles/DBoW2.dir/DBoW2/ScoringObject.cpp.o
[ 83%] Building CXX object CMakeFiles/DBoW2.dir/DUtils/Random.cpp.o
[100%] Building CXX object CMakeFiles/DBoW2.dir/DUtils/Timestamp.cpp.o
Linking CXX shared library ../lib/libDBoW2.so
/usr/bin/ld: cannot find -lQt5::Core
/usr/bin/ld: cannot find -lQt5::Gui
/usr/bin/ld: cannot find -lQt5::Widgets
/usr/bin/ld: cannot find -lQt5::Test
/usr/bin/ld: cannot find -lQt5::Concurrent
/usr/bin/ld: cannot find -lQt5::OpenGL
collect2: error: ld returned 1 exit status
make[2]: *** [../lib/libDBoW2.so] Error 1
make[1]: *** [CMakeFiles/DBoW2.dir/all] Error 2
make: *** [all] Error 2
Configuring and building Thirdparty/g2o ...
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- BUILD TYPE:Release
-- Compiling on Unix
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- A library with BLAS API found.
-- Looking for cheev_
-- Looking for cheev_ - found
-- A library with LAPACK API found.
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.1.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/roots/user1/ORB_SLAM2/Thirdparty/g2o/build
Scanning dependencies of target g2o
[  3%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_sba.cpp.o
[  6%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_six_dof_expmap.cpp.o
[ 10%] Building CXX object CMakeFiles/g2o.dir/g2o/types/types_seven_dof_expmap.cpp.o
[ 13%] Building CXX object CMakeFiles/g2o.dir/g2o/core/hyper_graph_action.cpp.o
[ 17%] Building CXX object CMakeFiles/g2o.dir/g2o/core/hyper_graph.cpp.o
[ 20%] Building CXX object CMakeFiles/g2o.dir/g2o/core/marginal_covariance_cholesky.cpp.o
[ 24%] Building CXX object CMakeFiles/g2o.dir/g2o/core/matrix_structure.cpp.o
[ 27%] Building CXX object CMakeFiles/g2o.dir/g2o/core/batch_stats.cpp.o
[ 31%] Building CXX object CMakeFiles/g2o.dir/g2o/core/parameter.cpp.o
[ 34%] Building CXX object CMakeFiles/g2o.dir/g2o/core/cache.cpp.o
[ 37%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimizable_graph.cpp.o
[ 41%] Building CXX object CMakeFiles/g2o.dir/g2o/core/solver.cpp.o
[ 44%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_factory.cpp.o
[ 48%] Building CXX object CMakeFiles/g2o.dir/g2o/core/estimate_propagator.cpp.o
[ 51%] Building CXX object CMakeFiles/g2o.dir/g2o/core/factory.cpp.o
[ 55%] Building CXX object CMakeFiles/g2o.dir/g2o/core/sparse_optimizer.cpp.o
[ 58%] Building CXX object CMakeFiles/g2o.dir/g2o/core/hyper_dijkstra.cpp.o
[ 62%] Building CXX object CMakeFiles/g2o.dir/g2o/core/parameter_container.cpp.o
[ 65%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm.cpp.o
[ 68%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_with_hessian.cpp.o
[ 72%] Building CXX object CMakeFiles/g2o.dir/g2o/core/optimization_algorithm_levenberg.cpp.o
[ 75%] Building CXX object CMakeFiles/g2o.dir/g2o/core/jacobian_workspace.cpp.o
[ 79%] Building CXX object CMakeFiles/g2o.dir/g2o/core/robust_kernel.cpp.o
[ 82%] Building CXX object CMakeFiles/g2o.dir/g2o/core/robust_kernel_factory.cpp.o
[ 86%] Building CXX object CMakeFiles/g2o.dir/g2o/core/robust_kernel_impl.cpp.o
[ 89%] Building CXX object CMakeFiles/g2o.dir/g2o/stuff/timeutil.cpp.o
[ 93%] Building C object CMakeFiles/g2o.dir/g2o/stuff/os_specific.c.o
[ 96%] Building CXX object CMakeFiles/g2o.dir/g2o/stuff/string_tools.cpp.o
[100%] Building CXX object CMakeFiles/g2o.dir/g2o/stuff/property.cpp.o
Linking CXX shared library ../lib/libg2o.so
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Found CUDA: /usr/local/cuda (found suitable exact version "7.5") 
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.1.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/roots/user1/ORB_SLAM2/build
Scanning dependencies of target ORB_SLAM2
[  4%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
[  8%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o
[ 13%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o
[ 17%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o
[ 21%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/ORBextractor.cc.o
[ 26%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/ORBmatcher.cc.o
[ 30%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o
[ 34%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Converter.cc.o
[ 39%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapPoint.cc.o
[ 43%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrame.cc.o
[ 47%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Map.cc.o
[ 52%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapDrawer.cc.o
[ 56%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
[ 60%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/PnPsolver.cc.o
[ 65%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o
[ 69%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrameDatabase.cc.o
[ 73%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Sim3Solver.cc.o
[ 78%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o
[ 82%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
make[2]: *** No rule to make target `../Thirdparty/DBoW2/lib/libDBoW2.so', needed by `../lib/libORB_SLAM2.so'.  Stop.
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2

Any help is much appreciated

kanster, 2016-08-18:
Hi @MCoder1 ,

The QT5 related dependencies are missing and caused the fail of DBoW library.

Try to install qt5 related dependencies from apt-get. I cannot find the exact package name, sorry you need to find it by yourself

Regards

MCoder1, 2016-08-18:
I resolved it by adding to the CMakeLists.txt for DBoW2 and ORB_SLAM2. 

find_package(Qt5Core)
find_package(Qt5Widgets)
etc.

closed 2016-08-18 by MCoder1