raulmur/ORB_SLAM2 #147: OS X Build Failure linking

adammw, 2016-08-19:
Building with CMake on OS X 10.11.6 with opencv and pangolin installed via Homebrew:

```
/usr/local/bin/g++-6   -Wall   -O3 -march=native -std=c++11 -O3 -DNDEBUG -dynamiclib -Wl,-headerpad_max_install_names  -o ../lib/libORB_SLAM2.dylib -install_name @rpath/libORB_SLAM2.dylib CMakeFiles/ORB_SLAM2.dir/src/System.cc.o CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o CMakeFiles/ORB_SLAM2.dir/src/ORBextractor.cc.o CMakeFiles/ORB_SLAM2.dir/src/ORBmatcher.cc.o CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o CMakeFiles/ORB_SLAM2.dir/src/Converter.cc.o CMakeFiles/ORB_SLAM2.dir/src/MapPoint.cc.o CMakeFiles/ORB_SLAM2.dir/src/KeyFrame.cc.o CMakeFiles/ORB_SLAM2.dir/src/Map.cc.o CMakeFiles/ORB_SLAM2.dir/src/MapDrawer.cc.o CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o CMakeFiles/ORB_SLAM2.dir/src/PnPsolver.cc.o CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o CMakeFiles/ORB_SLAM2.dir/src/KeyFrameDatabase.cc.o CMakeFiles/ORB_SLAM2.dir/src/Sim3Solver.cc.o CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o -F/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks /usr/local/lib/libopencv_videostab.2.4.13.dylib /usr/local/lib/libopencv_ts.a /usr/local/lib/libopencv_superres.2.4.13.dylib /usr/local/lib/libopencv_stitching.2.4.13.dylib /usr/local/lib/libopencv_contrib.2.4.13.dylib /usr/local/Cellar/pangolin/0.4/lib/libpangolin.dylib ../Thirdparty/DBoW2/lib/libDBoW2.dylib ../Thirdparty/g2o/lib/libg2o.dylib /usr/local/lib/libopencv_nonfree.2.4.13.dylib /usr/local/lib/libopencv_ocl.2.4.13.dylib /usr/local/lib/libopencv_gpu.2.4.13.dylib /usr/local/lib/libopencv_photo.2.4.13.dylib /usr/local/lib/libopencv_objdetect.2.4.13.dylib /usr/local/lib/libopencv_legacy.2.4.13.dylib /usr/local/lib/libopencv_video.2.4.13.dylib /usr/local/lib/libopencv_ml.2.4.13.dylib /usr/local/lib/libopencv_calib3d.2.4.13.dylib /usr/local/lib/libopencv_features2d.2.4.13.dylib /usr/local/lib/libopencv_highgui.2.4.13.dylib /usr/local/lib/libopencv_imgproc.2.4.13.dylib /usr/local/lib/libopencv_flann.2.4.13.dylib /usr/local/lib/libopencv_core.2.4.13.dylib -framework OpenCL -framework OpenGL /usr/local/lib/libGLEW.dylib -framework Cocoa /usr/lib/libpython2.7.dylib /usr/local/lib/libavcodec.dylib /usr/local/lib/libavformat.dylib /usr/local/lib/libavutil.dylib /usr/local/lib/libswscale.dylib /usr/local/lib/libpng.dylib /usr/lib/libz.dylib /usr/local/lib/libjpeg.dylib /usr/local/lib/libtiff.dylib /usr/local/lib/libIlmImf.dylib -Wl,-rpath,/usr/local/Cellar/pangolin/0.4/lib

Undefined symbols for architecture x86_64:
  "cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      ORB_SLAM2::System::System(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ORB_SLAM2::System::eSensor, bool) in System.cc.o
      ORB_SLAM2::Tracking::Tracking(ORB_SLAM2::System*, DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>*, ORB_SLAM2::FrameDrawer*, ORB_SLAM2::MapDrawer*, ORB_SLAM2::Map*, ORB_SLAM2::KeyFrameDatabase*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int) in Tracking.cc.o
      ORB_SLAM2::Tracking::ChangeCalibration(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) in Tracking.cc.o
      ORB_SLAM2::MapDrawer::MapDrawer(ORB_SLAM2::Map*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) in MapDrawer.cc.o
      ORB_SLAM2::Viewer::Viewer(ORB_SLAM2::System*, ORB_SLAM2::FrameDrawer*, ORB_SLAM2::MapDrawer*, ORB_SLAM2::Tracking*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) in Viewer.cc.o
  "cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)", referenced from:
      ORB_SLAM2::FrameDrawer::DrawTextInfo(cv::Mat&, int, cv::Mat&) in FrameDrawer.cc.o
  "cv::namedWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "cv::KeyPointsFilter::retainBest(std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, int)", referenced from:
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOld(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
  "cv::FAST(cv::_InputArray const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, int, bool)", referenced from:
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOctTree(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOld(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
  "cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double)", referenced from:
      DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::save(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const in System.cc.o
  "cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)", referenced from:
      cv::FileStorage& cv::operator<< <int>(cv::FileStorage&, int const&) in System.cc.o
      DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::save(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const in System.cc.o
  "cv::imshow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "cv::putText(cv::Mat&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)", referenced from:
      ORB_SLAM2::FrameDrawer::DrawTextInfo(cv::Mat&, int, cv::Mat&) in FrameDrawer.cc.o
  "cv::Exception::Exception(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)", referenced from:
      cv::FileStorage& cv::operator<< <int>(cv::FileStorage&, int const&) in System.cc.o
      DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::save(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const in System.cc.o
      cv::Mat::diag(cv::Mat const&) in Initializer.cc.o
  "cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      cv::operator<<(cv::FileStorage&, char const*) in System.cc.o
      DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::save(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const in System.cc.o
  "pangolin::CreatePanel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "pangolin::BindToContext(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)", referenced from:
      ORB_SLAM2::System::Shutdown()    in System.cc.o
  "pangolin::CreateWindowAndBind(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, pangolin::Params const&)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "pangolin::Split(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char)", referenced from:
[... cut ...]
  "cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const", referenced from:
      DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::load(cv::FileStorage const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) in System.cc.o
```

Been pulling my hair out trying trying to work out what's going wrong. Sample OpenCV tutorials compile fine, but I think it's using the static linking rather than dynamic linking.

I0x0I, 2016-08-26:
When you have issues with cxx11 and you are using gcc-5+, you might want to read [this page](https://gcc.gnu.org/onlinedocs/libstdc++/manual/using_dual_abi.html), you need to make sure that all the libraries you used have the same abi.

REIONE, 2016-10-16:
@adammw  Have you solved the issue?

derikon, 2017-04-26:
I experienced something similar.

- macOS Sierra 10.12.1
- OpenCV 3.2.0 (installed via Homebrew)
- Pangolin (built from source - master)

```
/usr/local/bin/g++-6   -Wall   -O3 -march=native -std=c++11 -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -dynamiclib -Wl,-headerpad_max_install_names  -o ../lib/libORB_SLAM2.dylib -install_name @rpath/libORB_SLAM2.dylib CMakeFiles/ORB_SLAM2.dir/src/System.cc.o CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o CMakeFiles/ORB_SLAM2.dir/src/ORBextractor.cc.o CMakeFiles/ORB_SLAM2.dir/src/ORBmatcher.cc.o CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o CMakeFiles/ORB_SLAM2.dir/src/Converter.cc.o CMakeFiles/ORB_SLAM2.dir/src/MapPoint.cc.o CMakeFiles/ORB_SLAM2.dir/src/KeyFrame.cc.o CMakeFiles/ORB_SLAM2.dir/src/Map.cc.o CMakeFiles/ORB_SLAM2.dir/src/MapDrawer.cc.o CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o CMakeFiles/ORB_SLAM2.dir/src/PnPsolver.cc.o CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o CMakeFiles/ORB_SLAM2.dir/src/KeyFrameDatabase.cc.o CMakeFiles/ORB_SLAM2.dir/src/Sim3Solver.cc.o CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o /usr/local/lib/libopencv_stitching.3.2.0.dylib /usr/local/lib/libopencv_superres.3.2.0.dylib /usr/local/lib/libopencv_videostab.3.2.0.dylib /usr/local/lib/libopencv_aruco.3.2.0.dylib /usr/local/lib/libopencv_bgsegm.3.2.0.dylib /usr/local/lib/libopencv_bioinspired.3.2.0.dylib /usr/local/lib/libopencv_ccalib.3.2.0.dylib /usr/local/lib/libopencv_dpm.3.2.0.dylib /usr/local/lib/libopencv_fuzzy.3.2.0.dylib /usr/local/lib/libopencv_hdf.3.2.0.dylib /usr/local/lib/libopencv_line_descriptor.3.2.0.dylib /usr/local/lib/libopencv_optflow.3.2.0.dylib /usr/local/lib/libopencv_reg.3.2.0.dylib /usr/local/lib/libopencv_saliency.3.2.0.dylib /usr/local/lib/libopencv_stereo.3.2.0.dylib /usr/local/lib/libopencv_structured_light.3.2.0.dylib /usr/local/lib/libopencv_surface_matching.3.2.0.dylib /usr/local/lib/libopencv_tracking.3.2.0.dylib /usr/local/lib/libopencv_xfeatures2d.3.2.0.dylib /usr/local/lib/libopencv_ximgproc.3.2.0.dylib /usr/local/lib/libopencv_xobjdetect.3.2.0.dylib /usr/local/lib/libopencv_xphoto.3.2.0.dylib /Volumes/Daten-I/STUDIUM/DIPLOMA/dev/3rd_party/dependencies/Pangolin/build/src/Release/libpangolin.dylib ../Thirdparty/DBoW2/lib/libDBoW2.dylib ../Thirdparty/g2o/lib/libg2o.dylib /usr/local/lib/libopencv_shape.3.2.0.dylib /usr/local/lib/libopencv_phase_unwrapping.3.2.0.dylib /usr/local/lib/libopencv_rgbd.3.2.0.dylib /usr/local/lib/libopencv_calib3d.3.2.0.dylib /usr/local/lib/libopencv_video.3.2.0.dylib /usr/local/lib/libopencv_datasets.3.2.0.dylib /usr/local/lib/libopencv_dnn.3.2.0.dylib /usr/local/lib/libopencv_face.3.2.0.dylib /usr/local/lib/libopencv_plot.3.2.0.dylib /usr/local/lib/libopencv_text.3.2.0.dylib /usr/local/lib/libopencv_features2d.3.2.0.dylib /usr/local/lib/libopencv_flann.3.2.0.dylib /usr/local/lib/libopencv_objdetect.3.2.0.dylib /usr/local/lib/libopencv_ml.3.2.0.dylib /usr/local/lib/libopencv_highgui.3.2.0.dylib /usr/local/lib/libopencv_photo.3.2.0.dylib /usr/local/lib/libopencv_videoio.3.2.0.dylib /usr/local/lib/libopencv_imgcodecs.3.2.0.dylib /usr/local/lib/libopencv_imgproc.3.2.0.dylib /usr/local/lib/libopencv_core.3.2.0.dylib -framework AGL -framework OpenGL /usr/local/lib/libGLEW.dylib -framework Cocoa /usr/lib/libpython2.7.dylib /usr/local/lib/libavcodec.dylib /usr/local/lib/libavformat.dylib /usr/local/lib/libavutil.dylib /usr/local/lib/libswscale.dylib /usr/local/lib/libpng.dylib /usr/lib/libz.dylib /usr/local/lib/libjpeg.dylib /usr/local/lib/libtiff.dylib /usr/local/lib/libIlmImf.dylib -Wl,-rpath,~/Pangolin/build/src/Release -Wl,-rpath,/usr/local/lib -Wl,-rpath,~/ORB_SLAM2/Thirdparty/DBoW2/lib -Wl,-rpath,~/ORB_SLAM2/Thirdparty/g2o/lib
```

results in this error:
```
Undefined symbols for architecture x86_64:
  "cv::KeyPointsFilter::retainBest(std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, int)", referenced from:
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOld(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
  "cv::FAST(cv::_InputArray const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, int, bool)", referenced from:
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOld(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOctTree(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
  "pangolin::CreatePanel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "pangolin::BindToContext(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)", referenced from:
      ORB_SLAM2::System::Shutdown()    in System.cc.o
  "pangolin::CreateWindowAndBind(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, pangolin::Params const&)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "pangolin::Split(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char)", referenced from:
      pangolin::Var<bool>::Var(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool const&, bool) in Viewer.cc.o
ld: symbol(s) not found for architecture x86_64
```

woooha, 2017-06-02:
I experienced same problem.

- macOS Sierra 10.12.4
- OpenCV 3.2.0(installed via Homebrew)
- Pangolin (built from source - master)

results in this error:

```
Undefined symbols for architecture x86_64:
  "cv::KeyPointsFilter::retainBest(std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, int)", referenced from:
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOld(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
  "cv::FAST(cv::_InputArray const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, int, bool)", referenced from:
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOld(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
      ORB_SLAM2::ORBextractor::ComputeKeyPointsOctTree(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) in ORBextractor.cc.o
  "pangolin::CreatePanel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "pangolin::BindToContext(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)", referenced from:
      ORB_SLAM2::System::Shutdown()    in System.cc.o
  "pangolin::CreateWindowAndBind(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int, pangolin::Params const&)", referenced from:
      ORB_SLAM2::Viewer::Run()    in Viewer.cc.o
  "pangolin::Split(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char)", referenced from:
      pangolin::Var<bool>::Var(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool const&, bool) in Viewer.cc.o
```

AndreasVaage, 2018-03-10:
@derikon, @woooha did any of you find a solution? I experience exactly the same problem.

armandok, 2018-04-18:
I am having the same problem :(

AndreasVaage, 2018-04-19:
@armandok I think my problem was somehow caused by using clang to compile some 3-party libraries and gcc to compile ORB-SLAM. I got it to work using https://github.com/kitizz/ORB_SLAM2, and modifying 
 include/LoopClosing.h line 49 and 50 from:
```
typedef map<KeyFrame*,g2o::Sim3,std::less<KeyFrame*>,
Eigen::aligned_allocator<std::pair<const KeyFrame*, g2o::Sim3> > > KeyFrameAndPose;
```
To:
```
typedef map<KeyFrame* const,g2o::Sim3,std::less<KeyFrame* const>,
Eigen::aligned_allocator<std::pair<KeyFrame* const, g2o::Sim3> > > KeyFrameAndPose;
```

And then compile it all using the built in clang on mac: Apple LLVM version 9.0.0 (clang-900.0.39.2). Hope this helps.

sgasperini, 2018-11-12:
@AndreasVaage how do you compile it all with clang?

when using clang I get errors with the "thirdparty" g2o that is looking for tr1/unordered_map (I installed boost with brew)

with gcc the "thirdparty" libraries build fine, but when building ORB_SLAM2 I get the same error from above, even with the Mac version from kitizz you linked

thank you!