raulmur/ORB_SLAM2 #156: Mono, ROS, Morse: Pangolin terminates with std::invalid_argument

artifactz, 2016-09-12:
When trying to use ORB_SLAM2 instead of ORB_SLAM (working fine) in our MORSE simulation, it crashes directly after initialization:

    terminate called after throwing an instance of 'std::invalid_argument'
      what():  'Look' and 'up' vectors cannot be parallel when calling ModelViewLookAt.
    [ORB_SLAM2-1] process has died [pid 32192, exit code -6, cmd /home/x/y/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/Mono Vocabulary/ORBvoc.txt Settings/Settings.yaml /camera/image_raw:=/robot/videocamera/image __name:=ORB_SLAM2  __log:=/home/x/.ros/log/21fe673c-78c7-11e6-a085-f0def123cb1d/ORB_SLAM2-1.log].

The caller is ORB_SLAM2/src/Viewer.cc. 'Look' and 'up' seem orthogonal though:

    pangolin::ModelViewLookAt(mViewpointX,mViewpointY,mViewpointZ, 0,0,0,0.0,-1.0, 0.0)

Does anyone have an idea what is going on? Any workarounds?

ebrand0007, 2016-10-03:
Having same issue.  Anyone figure this out?

ebrand0007, 2016-10-03:
Ok, dug through the code. You have to set the following parameters  Settings.yaml file. Not sure what they translate to yet..  Below is just example

Viewer.ViewpointX: 100
Viewer.ViewpointY: 100
Viewer.ViewpointZ: 100