raulmur/ORB_SLAM2 #159: A library with BLAS API found. However, a required library with LAPACK API not found.

libohit, 2016-09-23:
When I try to cmake g2o by following command, a CMake Error occurred. 
/ORB_SLAM2/Thirdparty/g2o/build$ cmake .. -DCMAKE_BUILD_TYPE=Release

I have installed blas and lapack by
sudo apt-get install libblas-dev
sudo apt-get install liblapack-dev
Can anybody help? I am struggling.

-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- BUILD TYPE:Release
-- Compiling on Unix
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- A library with BLAS API found.
-- Looking for cheev_
-- Looking for cheev_ - not found
-- Looking for cheev
-- Looking for cheev - not found
CMake Error at cmake_modules/FindLAPACK.cmake:247 (message):
  A required library with LAPACK API not found.  Please specify library
  location.
Call Stack (most recent call first):
  CMakeLists.txt:47 (FIND_PACKAGE)

M1234Thomas, 2016-10-07:
Please let me know if you have found a solution for this issue. I am facing this issue as well.

wystnphux, 2017-02-06:
Unfortunately, I'm having the same issue, but on Windows and with Visual Studio.
I downloaded LAPACK prebuilt from http://icl.cs.utk.edu/lapack-for-windows/lapack/#libraries_mingw.

Is this problem unrelated to the system on which orb slam is built?

buenos-dan, 2020-06-14:
I meet with the same problem in ubuntu 18.04, and I found a solution and fixed my problem.
apt-get install libatlas-base-dev libblas-dev liblapack-dev libhdf5-dev
firstly, I only install blas and lapack, it is not work, I think libatlas-base-dev is necessary.
reference link as follow: https://askubuntu.com/questions/1209605/a-required-library-with-lapack-api-not-found-please-specify-library-location