raulmur/ORB_SLAM2 #164: KITTI Dataset has 21 sequences, but in ReadMe example it is up to 11

Santi-7, 2016-09-26:
I have just downloaded the `KITTI Dataset` and have realized that there are **21** distinct sequences.

However, in `README.md` file it is shown how to play only the first **11** sequences. How do we play the rest of sequences (12-21)? I mean, which `.yaml` file should we use.

I have tested them with `KITTI04-12.yaml` and it seems to work correct.

SummerHuiZhang, 2019-04-24:
Hi~Have you found the yaml after 12? Could you please post the link?

SummerHuiZhang, 2019-04-24:
OK. I found it. Download data_odometry_calib.zip file in http://www.cvlibs.net/datasets/kitti/eval_odometry.php.
There is a calib.txt in each sequence. The first line is P0 (fx  0  cx  0  0  fy  cy  0  0  0   1  0 ). Here we get these 4 parameters of cameras.