raulmur/ORB_SLAM2 #167: What if I use unmodified DBoW2 ?

jiapei100, 2016-10-03:
I can see some difference of the file **TemplatedVocabulary.h** between both DBoW2

- ORB_SLAM2
- https://github.com/dorian3d/DBoW2

what if I insist on using the unchanged version DBoW2? and inherite the file **TemplatedVocabulary.h** ?

Cheers
Pei

jiapei100, 2016-10-06:
Still the same today...

> 
> /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int, long int>]’:
> /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:776:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int, long int>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
> /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:754:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>]’
> /usr/local/include/eigen3/Eigen/src/Core/PlainObjectBase.h:703:32:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Derived = Eigen::Matrix<int, -1, 1>]’
> /usr/local/include/eigen3/Eigen/src/Core/Matrix.h:225:24:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; _Scalar = int; int _Rows = -1; int _Cols = 1; int _Options = 0; int _MaxRows = -1; int _MaxCols = 1]’
> /usr/local/include/eigen3/Eigen/src/Core/PermutationMatrix.h:365:17:   required from ‘Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>& Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>::operator=(const Eigen::PermutationBase<OtherDerived>&) [with Other = Eigen::PermutationMatrix<-1, -1, long int>; int SizeAtCompileTime = -1; int MaxSizeAtCompileTime = -1; _StorageIndex = int]’
> /usr/local/include/g2o/solvers/linear_solver_eigen.h:68:18:   required from ‘void g2o::LinearSolverEigen<MatrixType>::CholeskyDecomposition::analyzePatternWithPermutation(g2o::LinearSolverEigen<MatrixType>::SparseMatrix&, const PermutationMatrix&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>; g2o::LinearSolverEigen<MatrixType>::SparseMatrix = Eigen::SparseMatrix<double, 0>; g2o::LinearSolverEigen<MatrixType>::PermutationMatrix = Eigen::PermutationMatrix<-1, -1, long int>]’
> /usr/local/include/g2o/solvers/linear_solver_eigen.h:195:9:   required from ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
> /usr/local/include/g2o/solvers/linear_solver_eigen.h:100:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, double*, double*) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
> /home/jiapei/Downloads/slam/OpenSLAM/ORB_SLAM2/src/Optimizer.cc:1244:1:   required from here
> /usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:811:3: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
>    EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
>    ^
> CMakeFiles/ORB_SLAM2.dir/build.make:353: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> make[2]: Leaving directory '/home/jiapei/Downloads/slam/OpenSLAM/ORB_SLAM2/build'
> CMakeFiles/Makefile2:70: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
> make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
> make[1]: Leaving directory '/home/jiapei/Downloads/slam/OpenSLAM/ORB_SLAM2/build'
> Makefile:86: recipe for target 'all' failed
> make: *** [all] Error 2
> 

Any suggestions please?
I didn't use the g2o nor DBoW2 coming with ORB_SLAM2. I used the 

- g2o from [https://github.com/RainerKuemmerle/g2o](https://github.com/RainerKuemmerle/g2o)
- DBoW2 from [https://github.com/dorian3d/DBoW2](https://github.com/dorian3d/DBoW2)

And I inherited

karolmajek, 2016-10-11:
Is there any way to compile this under Ubuntu 16.04?
I have the same error.
I have used provided  g2o and DBoW2.
@jiapei100 Did you solved this problem?

cd
git clone https://github.com/stevenlovegrove/Pangolin.git
cd Pangolin/
mkdir BUILD
cd BUILD
cmake .. -DCMAKE_BUILD_TYPE=Release
make -j 8
cd
git clone https://github.com/raulmur/ORB_SLAM2.git ORB_SLAM2
cd ORB_SLAM2
chmod +x build.sh
./build.sh

Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /home/karol/Pangolin/include/pangolin/gl/gl.h:40,
                 from /home/karol/Pangolin/include/pangolin/pangolin.h:34,
                 from /home/karol/ORB_SLAM2/include/MapDrawer.h:27,
                 from /home/karol/ORB_SLAM2/include/Viewer.h:26,
                 from /home/karol/ORB_SLAM2/include/Tracking.h:28,
                 from /home/karol/ORB_SLAM2/include/LocalMapping.h:27,
                 from /home/karol/ORB_SLAM2/include/LoopClosing.h:25,
                 from /home/karol/ORB_SLAM2/include/Optimizer.h:27,
                 from /home/karol/ORB_SLAM2/src/Optimizer.cc:21:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:712:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:693:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:682:32:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Derived = Eigen::Matrix<int, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:225:24:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; _Scalar = int; int _Rows = -1; int _Cols = 1; int _Options = 0; int _MaxRows = -1; int _MaxCols = 1]’
/usr/include/eigen3/Eigen/src/Core/PermutationMatrix.h:367:17:   required from ‘Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>& Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>::operator=(const Eigen::PermutationBase<OtherDerived>&) [with Other = Eigen::PermutationMatrix<-1, -1, long int>; int SizeAtCompileTime = -1; int MaxSizeAtCompileTime = -1; _StorageIndex = int]’
/home/karol/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:68:18:   required from ‘void g2o::LinearSolverEigen<MatrixType>::CholeskyDecomposition::analyzePatternWithPermutation(g2o::LinearSolverEigen<MatrixType>::SparseMatrix&, const PermutationMatrix&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>; g2o::LinearSolverEigen<MatrixType>::SparseMatrix = Eigen::SparseMatrix<double, 0>; g2o::LinearSolverEigen<MatrixType>::PermutationMatrix = Eigen::PermutationMatrix<-1, -1, long int>]’
/home/karol/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:195:9:   required from ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/karol/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:100:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, double*, double*) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/karol/ORB_SLAM2/src/Optimizer.cc:1244:1:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:707:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
   EIGEN_STATIC_ASSERT((internal::functor_is_product_like<BINOP>::ret \
   ^
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:745:3: note: in expansion of macro ‘EIGEN_CHECK_BINARY_COMPATIBILIY’
   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
   ^
CMakeFiles/ORB_SLAM2.dir/build.make:350: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

geokarant, 2016-10-13:
Hello, i faced a similar problem.. Try the latest version of Eigen.  I used Eigen 3.3-rc1 release! For openCV i also installed the 2.4.13 version. I wish you good luck.

jiapei100, 2016-10-20:
Problem solved... It seems you have to enable C++11, and have to use the Thirdparty g2o and DBoW2, instead of original g2o nor DBoW2...

Thanks...

closed 2016-10-20 by jiapei100