raulmur/ORB_SLAM2 #169: super real-time possible?

nrhine1, 2016-10-04:
Could straightforward modifications be made to speed the algorithm up to run faster than real time (from a pre-recorded video)? I've removed the `cv::waitKey` in Viewer.cc (doing my own event processing), reduced the number of ORB features per frame, and have been tracking down dependencies on the input framerate, but haven't achieved faster than real-time performance yet (using non-ROS version)

thanks

AlejandroSilvestri, 2016-10-18:
Yes, it is possible with enough processor speed.
Removing waitKey is the right thing to do.
It is also very dependent on your video resolution an fps: 120 fps is 4 times time consuming than 30 fps :)

aivijay, 2016-10-21:
How are you reducing the features per frame? Is it possible that setting Camera.width and Camera.height in the calibration file control the frame size?