raulmur/ORB_SLAM2 #171: g2o is NOT up to date....

jiapei100, 2016-10-06:
Just wonder if the most up-to-date g2o [https://github.com/dorian3d/DBoW2](https://github.com/dorian3d/DBoW2) is also compatible with ORB_SLAM2?

Cheers
Pei

AlejandroSilvestri, 2016-10-18:
Both thirdparty packages (g2o or DBoW2) were minimized and modified to be included in orb-slam2.

I think someone has to try to compile orb-slam2 with up to date g2o and/or dbow2 and see what happens.

The other question: has new g2o version improved some code orb-slam2 uses?

jiapei100, 2016-10-20:
I've already figure it out by myself... 

Thank you ...

closed 2016-10-20 by jiapei100

AlejandroSilvestri, 2016-10-20:
Excellent! So, it's recommended to patch BaseBinaryEdge.
Thank you!

gzhang8, 2017-08-30:
I got a problem when using official latest g2o with orb-slam2. To whom does not feel @jiapei100 's solution that clear. Here is a detailed version of how I solved this problem.

Replacing 

types_six_dof_expmap.h
types_six_dof_expmap.cpp
types_seven_dof_expmap.h
types_seven_dof_expmap.cpp

in latest g2o with ones from g2o in this repo

Then fix headers problem in these four files, also disable building examples and apps. In addition, use findG2O instead of using one libg2o.so

Everything in ORB-SLAM2 with work fine with it.

vigorbird, 2018-03-14:
The ORB author said that the DBow and g2o is the modified version. I want to know that the modified version g2o and DBow is just part of original g2o and DBow code or the ORB author changes some codes in original g2o and DBow codes. If the author changes the codes , why he does so?

vigorbird, 2018-03-14:
@AlejandroSilvestri,@AlejandroSilvestri
The ORB author said that the DBow and g2o is the modified version. I want to know that the modified version g2o and DBow is just part of original g2o and DBow code or the ORB author changes some codes in original g2o and DBow codes. If the author changes the codes , why he does so?

AlejandroSilvestri, 2018-03-14:
@vigorbird ,

I don't remember clearly, so I can't cite the source, but I quite sure the only modification was deleting parts orb-slam2 don't use.

vigorbird, 2018-03-14:
@AlejandroSilvestri 
Thank you very much. I just finish reading orb2 source code and find this problem. If I know what the reason why the author does so, I will finish this question.

Shashika007, 2019-05-17:
@gzhang8 @AlejandroSilvestri . I did the similar things you mentioned to upgrade newest g2o. But, i ran into some other issues for following,
sparse_block_matrix_ccs.h - error: ‘number_t’ was not declared in this scope
and similar errors in other files also, Should i replace those files also ? 

If i replace those, would that be effective ?

thanks. well appreciated!