raulmur/ORB_SLAM2 #174: What is coordinate system used by ORB SLAM to represent map points?

Zhang-Yun, 2016-10-15:
Hi ,sorry for the entry-level question,I am wondering what kind of coordinate system it used for map points? the unit of this coordinate system?Hope someone can tell me,Thank you very much.

geokarant, 2016-10-17:
I am wondering the same... Where in the code can we understand the origin of the coordinate system used.. Thanks in advance!

AlejandroSilvestri, 2016-10-18:
Hi, the coordinate system is that of the first frame camera pose in first succeded triangulation.
There's no way monocular SLAM can know the real world scale, so it uses an arbitrary one.  Not sure, but I think the unit is the separation (translation) between the two camera poses taking for triangulation during initialization.

geokarant, 2016-10-18:
@AlejandroSilvestri  thank you very much for your response! ONe more question, in TRajectory.txt  what does each column represent? I cant figure it out.. 

Thanks in advance!

AlejandroSilvestri, 2016-10-18:
You are welcome.

There are two methods to save trajectory in [System.cc](https://github.com/raulmur/ORB_SLAM2/blob/master/src/System.cc):

System::SaveTrajectoryTUM
At line 367 you can tell the meaning of each column.

System::SaveTrajectoryKITTI
Same as before, at line 418.

geokarant, 2016-10-18:
thank you very much alejandro... i am new to slam and i am carrying out my
master thesis... I thank you a lot!

geokarant, 2016-10-18:
Thank you very much Alejandro! You helped me a lot!

Zhang-Yun, 2016-10-24:
@AlejandroSilvestri thank you for your reply,it helps me a lot.
I have found the following code in Initializer::DecomposeE():
void Initializer::DecomposeE(const cv::Mat &E, cv::Mat &R1, cv::Mat &R2, cv::Mat &t)
{
    cv::Mat u,w,vt;
    cv::SVD::compute(E,w,u,vt);

    u.col(2).copyTo(t);
    **t=t/cv::norm(t);**//Does this line set the unit transition for monocular SLAM that you mentioned?
   ........
}
Does t=t/cv::norm(t) set the unit transition for monocular SLAM that you mentioned?
Thank you.

AlejandroSilvestri, 2016-10-25:
Yes, that's it!

You should know t module is useless at this step, so it's cleaner to make it be 1.

Initializer get the essential matrix E from the two frames' poses.  DecomposeE get rotation R and translation t from E.

Zhang-Yun, 2016-10-25:
@AlejandroSilvestri thank you for your response.

AlejandroSilvestri, 2016-10-28:
Yes, you're right!  It normalize the map based on the scene median depth.
It seems very difficult to predict the measurement unit in a world to be initialized.

closed 2016-10-28 by Zhang-Yun

salsicha, 2017-07-12:
Hi again, I've run stereo slam (zed camera) and saved the trajectory. I see that the trajectory data is from a matrix but I don't know what columns in the trajectory file to plot in order to reconstruct the path of the camera. What do the values in the matrix or trajectory file represent? Is it as simple as x,y,z displacement values from an origin? Thanks!

rishshah, 2018-01-16:
8 columns 
timestamp  camera position (x, y, z) camera orientation in quarternion(q1 , q2, q3, q4)

Roshnee, 2019-07-01:
It is of the form [R00, R01, R02, t0, R10, R11, R12, t1, R20, R21, R22, t2] for the KITTI trajectory. 
Someone please correct me if I am wrong.

Roshnee, 2019-07-01:
> Hi again, I've run stereo slam (zed camera) and saved the KITTI trajectory. I see that the trajectory data is from a matrix but I don't know what columns in the trajectory file to plot in order to reconstruct the path of the camera. What do the values in the matrix or trajectory file represent? Is it as simple as x,y,z displacement values from an origin? Which column is x, which is y? Thanks!

It is of the form [R00, R01, R02, t0, R10, R11, R12, t1, R20, R21, R22, t2] for the KITTI trajectory.
Someone please correct me if I am wrong.