raulmur/ORB_SLAM2 #181: Replace PATH_TO_SEQUENCE_FOLDER with a real camera, without ROS

jiapei100, 2016-10-24:
Is there a way to run ORB_SLAM2 on a real camera without using that ROS demo?

Cheers
Pei

poine, 2016-10-25:
It depends on your operating system and on your real camera, I guess...
For example, if you are running Linux and a compatible camera, you could
use v4l and opencv to grab images and feed them to orbslam2.
hth
Poine

AlejandroSilvestri, 2016-10-27:
@poine is right, I'm doing it for a while now, capturing webcam with opencv and feeding orb-slam2 without ROS.  It works pretty well, as expected.  Don't forget to calibrate your webcam.

Of course, you have to modifiy some c++ code to do it, then compile.

Jephinus, 2017-08-23:
Did anyone resolve this issue? I have no clue yet