raulmur/ORB_SLAM2 #185: ORB Slam2

venkataramanas1, 2016-10-27:
Hi all ,

I  have built  ORB Slam2 project , the open source code and the output is not deterministic .  It gives random output for every iteration.

Is it because of random functions that are used ?  . Can you guys help me to fix this by guiding me to identify potential functions which may cause random behaviour.

AlejandroSilvestri, 2016-10-27:
RANSAC uses random variables.  And RANSAC is used in many methods in orb-slam2.

I don't think making a deterministic orb-slam is a good idea.

mojovski, 2016-10-27:
@AlejandroSilvestri  depends :) a fixed seed would help to evaluate the quality in continuous benchmarks.

venkataramanas1, 2016-10-28:
thanks for the reply

closed 2016-10-28 by venkataramanas1

AlejandroSilvestri, 2016-10-28:
@mojovski, you are right!

I'm not sure if doing so is an easy or near impossible task, but it worth trying, or at least looking for RANSAC in orb-slam2 and g2o.