raulmur/ORB_SLAM2 #186: OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName

fanyuzeng, 2016-11-01:
I used kinect v1.  
cd ORB-SLAM2
rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt  Examples/RGB-D/TUM1.yaml

but reslut in such error below:
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /home/zfy/Desktop/studySoft/opencv-2.4.11/modules/core/src/persistence.cpp, line 740
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/zfy/Desktop/studySoft/opencv-2.4.11/modules/core/src/persistence.cpp:740: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName

Aborted (core dumped)

How to deal with that.

geokarant, 2016-11-01:
you didnt provide any image sequence as i can figure out.  You can either
run a sequence without ros or if you want to use ros you should provide a
bag file.. Take a look at the examples in orb-slam (first version) and then
in orb-slam 2!

fanyuzeng, 2016-11-02:
Thank you. I solved the problem.
CMakeLists.txt 
find_package(OpenCV xxx REQUIRED)
I made the version xxx be consistent with the version of installed opencv, and it worked.

fanyuzeng, 2016-11-02:
@geokarant  Thank you. I solved the problem.
CMakeLists.txt 
find_package(OpenCV xxx REQUIRED)
I made the version xxx be consistent with the version of installed opencv, and it worked.

cmxnono, 2016-11-14:
Same with https://github.com/raulmur/ORB_SLAM2/issues/103#issuecomment-260386195

closed 2017-02-17 by fanyuzeng