raulmur/ORB_SLAM2 #189: planar motion: 6DoF or 3DoF pose

dannyway03, 2016-11-03:
Hi all,

never tired to thank you @raulmur for making all this available!
I know the framework already achieves remarkable performance in many datasets, but i was wondering if switching from current 6DoF (R[3x3], T[3x1]) keyframe pose model to 3DoF(T[2x1], Theta[1x1]) would be beneficial by any mean (speed, accuracy, ..) while assuming a locally planar motion (e.g., robot moving on a flat surface).

Anybody has some clues on how i could try this, or can redirect me to relevant parts of the code?
thank you in advance,
d

jingpang, 2016-11-07:
There may be one trick, to set the terms VERY LARGE in information matrix corresponding to 3 'fixed' DOF. If you'd like to make sure they not change at all, maybe you should rewrite the g2o vertex/edge.

AlejandroSilvestri, 2016-11-13:
Motion model will automatically "adapt" to 3DoF: when you move on a flat surface its forecast will naturally be in 3DoF.

Perhaps you can narrow matching region in tracking, reducing matching effort, but it is already optimized, I don't see a great improvement opportunity there.

    ORBmatcher::SearchByProjection(...)

In real situations, your robot will vibrate, oscilate, bump and jump (little jumps), so strictly speaking you can't abandon 6 DoF.