raulmur/ORB_SLAM2 #190: ORB_SLAM2 runs but no camera data ingestion

jacksonkr, 2016-11-04:
Ubuntu 14.04 | 4.4.0-46-generic
ORB_SLAM2 master

I've tried both monocular and rgbd with the same results. The app starts and I see a black screen with "waiting for images".

Screenshot:
http://imgur.com/a/1T01c

bash commands:
```
$ rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml 

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.09294

```

geokarant, 2016-11-04:
You may have to provide a .bag file. See the explanations in Orb slam v.1

jacksonkr, 2016-11-04:
@geokarant A bag is a ros recording. Is orb slam 2 strictly offline only? I was under the impression that you could use it in realtime.

geokarant, 2016-11-04:
You are right, of course it s for real time but I haven't tried it yet! I
think there is a webcam ros package or something.. I f you figure it out
let me know, please! Thank you in advance!!

jacksonkr, 2016-11-04:
I'm using an Intel RealSense r200. The package for that is `ros-[distro]-realsense_camera`. I'll give you an example with a kinect though:

```
sudo apt-get install ros-[distro]-openni-* # or freenect depending on which system you want
roscore & # you'll need ros installed, I recommend indigo http://wiki.ros.org/indigo/Installation
roslaunch openni_launch openni.launch # or freenect
rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml
```

That SHOULD be all but something must be missing because you won't see the live camera feed, you'll see that the slam is "waiting for images"

geokarant, 2016-11-04:
Thank you very much, I ll try it too, and let you know!

jacksonkr, 2016-11-05:
I tried orbslam2 with the kinect v1 and it worked fine so I decided it was a topic naming issue, which it was.

The depth topic orb slam 2 is looking for is `/camera/depth_registered/image_raw` and the `freenect_launch` specifies this whereas the `r200_nodelet` does not, it instead maps depth to `/camera/depth/image_raw`. You can change the launch files themselves if you want but the easiest way to get going is by running the following:

`$ rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml /camera/depth_registered/image_raw:=/camera/depth/image_raw`

closed 2016-11-05 by jacksonkr

Balaji-Govindaraj, 2016-11-25:
how to use usb camera for ros examples?

geokarant, 2016-11-25:
Search on ros wiki for webcam_node for ros. I havent tried it but i know it
exists

farhanmubasher, 2019-01-01:
> I tried orbslam2 with the kinect v1 and it worked fine so I decided it was a topic naming issue, which it was.
>
> The depth topic orb slam 2 is looking for is `/camera/depth_registered/image_raw` and [...]

HI @jacksonkr ,
I'm facing same problem as you were that orb-slam2 is running and showing blank screen "Waiting for images",
I don't know how to run it with real time, what commands and launch files i have to change. I'm using kinetic version of ROS, and my camera topics are `/stereo/left/image_raw` and  `/stereo/right/image_raw.`

 I have no idea of `freenect_launch`,
 Please help me how can i run it with real time. I will appreciate your time.

jacksonkr, 2019-01-01:
@farhanmubasher This was a while ago so I can't quite remember the details. Either orb slam wouldn't work or I couldn't get it to do what I needed.

As far as `freenect_launch` is concerned it's part of the ros freenect package (ros-freenect / libfreenect) which can be used with the kinect v1.

Good luck!