raulmur/ORB_SLAM2 #193: Error while building ORB SLAM 2 with @MathewDenny 's fork

francotestori, 2016-11-10:
Hey I am trying to build ORB SLAM 2 based on the fork from @MathewDenny (https://github.com/MathewDenny/ORB_SLAM2). 

I have already fixed some issues, like adding "tools/bin_vocabulary.cc" from @poine fork (check https://github.com/poine/ORB_SLAM2/blob/master/tools/bin_vocabulary.cc) and also added `#include <iomanip>` to both mono_kitti.cc and stereo_kitti.cc.

Nonetheless I can't successfully build ORB SLAM 2. I would appreciate any guidance in the subject.

Here I leave you my build output:

> $ ./build.sh 
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/franco/devel/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/franco/devel/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   serialization
-- Configuring done
-- Generating done
-- Build files have been written to: /home/franco/devel/ORB_SLAM2/build
[ 66%] Built target ORB_SLAM2
[ 70%] Linking CXX executable ../tools/bin_vocabulary
[ 73%] Linking CXX executable ../Examples/RGB-D/rgbd_tum
[ 76%] Linking CXX executable ../Examples/Monocular/mono_kitti
[ 80%] Linking CXX executable ../Examples/Stereo/stereo_kitti
/usr/bin/ld: CMakeFiles/bin_vocabulary.dir/tools/bin_vocabulary.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/opt/ros/kinetic/lib/libopencv_core3.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/bin_vocabulary.dir/build.make:136: recipe for target '../tools/bin_vocabulary' failed
make[2]: *** [../tools/bin_vocabulary] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/bin_vocabulary.dir/all' failed
make[1]: *** [CMakeFiles/bin_vocabulary.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/opt/ros/kinetic/lib/libopencv_core3.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/mono_kitti.dir/build.make:136: recipe for target '../Examples/Monocular/mono_kitti' failed
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
CMakeFiles/Makefile2:215: recipe for target 'CMakeFiles/mono_kitti.dir/all' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
/usr/bin/ld: CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/opt/ros/kinetic/lib/libopencv_core3.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_kitti.dir/build.make:136: recipe for target '../Examples/Stereo/stereo_kitti' failed
make[2]: *** [../Examples/Stereo/stereo_kitti] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/stereo_kitti.dir/all' failed
make[1]: *** [CMakeFiles/stereo_kitti.dir/all] Error 2
/usr/bin/ld: CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/opt/ros/kinetic/lib/libopencv_core3.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_tum.dir/build.make:136: recipe for target '../Examples/RGB-D/rgbd_tum' failed
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
Converting vocabulary to binary
./build.sh: line 35: ./tools/bin_vocabulary: No such file or directory

francotestori, 2016-11-10:
Might it be related with Open CV version?

closed 2016-11-17 by francotestori

linpeisensh, 2020-12-09:
hello,

I have the same problem. Could you tell me, whether it works with only opencv3.2 (I have used opencv3.4 and this error occurs)?

Thank you!