raulmur/ORB_SLAM2 #196: colored map

wmy14038tony, 2016-11-17:
How can we get the colored map as shown in the new youtube video? 
It seems that we can only get feature points with no color information. 
Besides, are those colored points all ORB key points?

AlejandroSilvestri, 2016-11-18:
Are you talking about these video?
https://youtu.be/ufvPS5wJAx0?t=2m2s

No, those aren't keypoints, those are rgb-d.

demplo, 2016-11-21:
[This one](https://www.youtube.com/watch?v=HlBmq70LKrQ) comes from Mono though.
The process for obtaining that result is described [here](http://www.roboticsproceedings.org/rss11/p41.pdf)

callsty, 2016-12-21:
It's very interesting but it doesn't explain how to run orb-slam2 with semi dense reconstruction. It's not activate by default.

ank700, 2017-02-05:
Hello, regarding the sparse map generated by orb-slam2, what do the black and red map points represent? I dont see any difference in the DrawMapPoints function at [https://github.com/raulmur/ORB_SLAM2/blob/132d92fc86ced1926c00509337fd40b445d9e29c/src/MapDrawer.cc#L44](url)

AlejandroSilvestri, 2017-02-05:
See lines 46 and 47.  Reference map points will be shown in red, the rest in black.

Reference map point are the ones corresponding to local map, the ones that are "in use".  Tracking ignores the rest.

https://github.com/raulmur/ORB_SLAM2/blob/master/src/MapDrawer.cc#L44