raulmur/ORB_SLAM2 #199: Localization concept

BingyuZhou, 2016-11-22:
Hi, 

Does anyone know what kind of localization method is used in the ORB-SLAM2 localization mode?

Thanks a lot!

AlejandroSilvestri, 2016-11-24:
Hi, I don't know exactly what are you expecting as an answer, but I can tell it uses tracking (Structure From Motion like) and relocalization in localization mode.

This mode deactivate the mapping, the thread responsible for adding new 3D points to the map and bundle adjusting them, and the thread responsible for loop closing, so the map remains unchanged.

BingyuZhou, 2016-11-29:
Hi, AlejandroSilvestri,

Thanks for your reply.

 I have read the ORB-SLAM2 paper. I understand that:  the pose of the camera can be obtained from the bundle adjustment procedure during the SLAM process and the pose can be obtained from the matching of existing map features or the visual odometry matching. But I am wondering how accurate is this localization. In EKF or Monte Carlo localization approaches, we can get the probability distribution of the pose. What about in ORB-SLAM? Is that possible to get the covariance or distribution of the estimated pose?

AlejandroSilvestri, 2016-11-29:
ORB-SLAM belong to "non-filter based visual SLAM" category.  It doesn't use covariance matrix nor EKF,  nor any variant or those.  Instead it rely on bundle adjustment to minimize error in map points positions.

Perhaps you could use the reprojection error, the difference between the projected map point onto the camera, and the actual 2D keypoint position.

BingyuZhou, 2016-11-30:
Thanks a lot! It is bit clearer now!

closed 2016-11-30 by BingyuZhou

lokeshluckybisht, 2018-06-21:
hi @BingyuZhou ,

I also want to get covariance from ORB_SLAM2. Did you find a method to extract it? 
Any help is appreciated.