raulmur/ORB_SLAM2 #201: Pangolin (?) Issues with ORB SLAM

passion-creativity-enthusiasm, 2016-11-23:
Hi enthusiasts,

I have an issue with ORB SLAM non-ros version in recent days that really was rustrating. I am running on Ubuntu 16.04 with Nvidia GTX950 cardand for some reason whenever I run my ORB SLAM with a dataset of 17545 files, the system will hang abruptly citing **segmentation fault (core dump)**.

There is an initial message stating, "**Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.**"

I have followed the instructions as kindly written in the documentation and previously I have setup ORB SLAM to run smoothly in Ubuntu 14.04 with an AMD card.

passion-creativity-enthusiasm, 2016-11-26:
In case anyone encounter this problem again, unfortunately, it can be resolved by re-installing in Ubuntu 14.04 :)

Thanks all!

closed 2016-11-26 by passion-creativity-enthusiasm

cumtchenchang, 2017-12-17:
I encounter the same issue, can it be solved except re-installing ubuntu

xujinjinjinjin, 2020-12-18:
I encounter the same issue with ORB SLAM ros version,can it be solved except re-installing ubuntu？

xujinjinjinjin, 2020-12-20:
> I encounter the same issue with ORB SLAM ros version,can it be solved except re-installing ubuntu？

I bypassed it by going in the g2o cmakelists and the orb slam cmakelists and removing -march=native .yeah~