raulmur/ORB_SLAM2 #202: Failed to build ORB-SLAM2 UBUNTU 16.04

haythemsgh, 2016-11-23:
Hi,
I want to install Orbslam2 on ubuntu 16.04.
I've installed all the necessary dependencies. At first i've got error to build the third library Dbow2 because i've installed opencv 3.1 then i've fixed by installing opencv 2.4.11 and modified the cmakelists by replacing find_package(opencv required) by opencv 2.4.11 and it worked. But when it comes to building orbslam it stopped at 81% and show me this message.

[image]
[image]
[image]

drinking-tea, 2016-11-24:
@haythemsgh I don't have this problem but I did meet other problems during installation. I solved them by adding `cmake -D OpenCV_DIR=/usr/local/opencv-2.4.13/release/ ..` to the `build.sh` file (http://stackoverflow.com/questions/18327197/opencv-is-considered-to-be-not-found). I am not sure if this helps but worth a try.

mmattamala, 2016-11-25:
Another solution to make it work with OpenCV 3 is to delete the 2.4 dependency in this line: https://github.com/raulmur/ORB_SLAM2/blob/master/CMakeLists.txt#L31

I tried it in Ubuntu 16.04 with ROS Kinetic (which includes OpenCV 3) and it worked without problems.

haythemsgh, 2016-11-26:
@drinking-tea  Thank you for your response, but the thing is that I want to ORBslam2 with ROS Kinetic so I want to build it with opencv 3.1.
@mmattamala I deleted the dependency of 2.4 and changed with find_package(OpenCV 3.1.0 REQUIRED) I also added this line in Line 2 set(OpenCV_DIR=/opt/ros/kinetic/share/OpenCV-3.1.0-dev) ("That's the directory of opencvconfigversion.cmake") but when I run make -j I get this new error.

In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /home/haythem/Pangolin/include/pangolin/gl/gl.h:39,
                 from /home/haythem/Pangolin/include/pangolin/pangolin.h:33,
                 from /home/haythem/ORB_SLAM2/include/MapDrawer.h:27,
                 from /home/haythem/ORB_SLAM2/include/Viewer.h:26,
                 from /home/haythem/ORB_SLAM2/include/Tracking.h:28,
                 from /home/haythem/ORB_SLAM2/include/LocalMapping.h:27,
                 from /home/haythem/ORB_SLAM2/include/LoopClosing.h:25,
                 from /home/haythem/ORB_SLAM2/include/Optimizer.h:27,
                 from /home/haythem/ORB_SLAM2/src/Optimizer.cc:21:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:712:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:693:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:682:32:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Derived = Eigen::Matrix<int, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:225:24:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; _Scalar = int; int _Rows = -1; int _Cols = 1; int _Options = 0; int _MaxRows = -1; int _MaxCols = 1]’
/usr/include/eigen3/Eigen/src/Core/PermutationMatrix.h:367:17:   required from ‘Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>& Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>::operator=(const Eigen::PermutationBase<OtherDerived>&) [with Other = Eigen::PermutationMatrix<-1, -1, long int>; int SizeAtCompileTime = -1; int MaxSizeAtCompileTime = -1; _StorageIndex = int]’
/home/haythem/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:68:18:   required from ‘void g2o::LinearSolverEigen<MatrixType>::CholeskyDecomposition::analyzePatternWithPermutation(g2o::LinearSolverEigen<MatrixType>::SparseMatrix&, const PermutationMatrix&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>; g2o::LinearSolverEigen<MatrixType>::SparseMatrix = Eigen::SparseMatrix<double, 0>; g2o::LinearSolverEigen<MatrixType>::PermutationMatrix = Eigen::PermutationMatrix<-1, -1, long int>]’
/home/haythem/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:195:9:   required from ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/haythem/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:100:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, double*, double*) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/haythem/ORB_SLAM2/src/Optimizer.cc:1244:1:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:707:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
   EIGEN_STATIC_ASSERT((internal::functor_is_product_like<BINOP>::ret \
   ^
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:745:3: note: in expansion of macro ‘EIGEN_CHECK_BINARY_COMPATIBILIY’
   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
   ^
CMakeFiles/ORB_SLAM2.dir/build.make:350: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I've tried another solution to change SparseMatrix::Index to int in  ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h but nothing worked.

cranil, 2016-11-28:
> I've tried another solution to change SparseMatrix::Index to int in ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h but nothing worked.

Try doing the opposite. I fixed it by changing `typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, SparseMatrix::Index> PermutationMatrix;`
to `typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, SparseMatrix::StorageIndex> PermutationMatrix;`

haythemsgh, 2016-11-28:
@cranil Thank you very much .Thanks to your solution I succeeded to build orbslam2 in ubuntu 16.04 without errors.

escrowdis, 2017-07-06:
The solution of @drinking-tea is a way to solve this! The true problem of this error is link to the wrong Eigen lib directory. Maybe your system has installed more than one eigen with another version before? so that the newer version doesn't have 
```
typeddef ... SparseMatrix::Index...
```
You can edit **CMakeLists.txt** to set `EIGEN3_INCLUDE_DIR` to direct to the right dir.

ujjwalrathod007, 2018-06-22:
I also have similar issue I am attaching a file which contains my bash output. Can someone suggest a fix?

[Untitled Document.tar.gz](https://github.com/raulmur/ORB_SLAM2/files/2128972/Untitled.Document.tar.gz)

shumshersubashgautam, 2020-06-27:
I stuck on this error any idea on this error:

/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_euroc.dir/build.make:165: recipe for target '../Examples/Monocular/mono_euroc' failed
make[2]: *** [../Examples/Monocular/mono_euroc] Error 1
CMakeFiles/Makefile2:294: recipe for target 'CMakeFiles/mono_euroc.dir/all' failed
make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_tum.dir/build.make:165: recipe for target '../Examples/Monocular/mono_tum' failed
make[2]: *** [../Examples/Monocular/mono_tum] Error 1
CMakeFiles/Makefile2:220: recipe for target 'CMakeFiles/mono_tum.dir/all' failed
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_tum.dir/build.make:165: recipe for target '../Examples/RGB-D/rgbd_tum' failed
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
CMakeFiles/Makefile2:109: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_kitti.dir/build.make:165: recipe for target '../Examples/Monocular/mono_kitti' failed
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
CMakeFiles/Makefile2:257: recipe for target 'CMakeFiles/mono_kitti.dir/all' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_kitti.dir/build.make:165: recipe for target '../Examples/Stereo/stereo_kitti' failed
make[2]: *** [../Examples/Stereo/stereo_kitti] Error 1
CMakeFiles/Makefile2:146: recipe for target 'CMakeFiles/stereo_kitti.dir/all' failed
make[1]: *** [CMakeFiles/stereo_kitti.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_euroc.dir/build.make:165: recipe for target '../Examples/Stereo/stereo_euroc' failed
make[2]: *** [../Examples/Stereo/stereo_euroc] Error 1
CMakeFiles/Makefile2:183: recipe for target 'CMakeFiles/stereo_euroc.dir/all' failed
make[1]: *** [CMakeFiles/stereo_euroc.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

robertolisandro, 2022-05-26:
> I stuck on this error any idea on this error:
>
> /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current' collect2: error: ld returned 1 exit [...]

I got a very similar error with TUM Dataset

Shengshu-Liu, 2022-07-04:
I have the same issue

- [ ] a

- [ ] b