raulmur/ORB_SLAM2 #204: ros examples

Balaji-Govindaraj, 2016-11-25:
When running Ros examples for monocular camera black screen appears with a text waiting for images

geokarant, 2016-11-25:
you need to provide an example package with images for ros. Otherwise, run
webcam node in order to stream online!

Balaji-Govindaraj, 2016-11-25:
Even if i run webcam am facing the same issue

Balaji-Govindaraj, 2016-11-25:
should i change any parameters in ros_ mono .cc?

geokarant, 2016-11-25:
i dont know, because i have never used it before. i hope someone will help.

hari0920, 2017-01-20:
I am facing the same problem. Can anyone provide some information on this issue??

ThomAub, 2018-04-20:
Hi all, 
Can you check your rostopic list ? 
orb-slam take as input "camera/image-raw" maybe you're not streaming images on the right topic ? 

best,

zhkmxx9302013, 2018-06-27:
If you use the usb_cam, you need to change the default topic of the orbslam2.
As the usb_cam's default topic is 'usb_cam', while the orbslam2 ROS monocular example's default topic is 'camera'.
Open the 'Examples/ROS/ORB_SLAM2/src/ros_mono.cc ', modify the subscribe topic to usb_cam, `ros::Subscriber sub = nodeHandler.subscribe("/usb_cam/image_raw", 1, &ImageGrabber::GrabImage,&igb);` then rebuild the example.

reference:
https://blog.csdn.net/goding_learning/article/details/52950993

ujjwalrathod007, 2018-09-08:
So the above discussion concludes that if one has USB camera he needs to publish the images from the camera and read them as an input to the node? 
Do I also need to edit ros_mono.cc file?
Anyone who has tried may have an idea..

zoldaten, 2022-09-12:
@ujur007 
u dont need do both remap and change ros_mono.cc. 
do what you want:
**change ros_mono.cc**:
`ros::Subscriber sub = nodeHandler.subscribe("/camera/image_raw", 1, &ImageGrabber::GrabImage,&igb);`
to 
`ros::Subscriber sub = nodeHandler.subscribe("/usb_camera/image_raw", 1, &ImageGrabber::GrabImage,&igb);`
and rebuild examples
**alternative:**
in usb_cam launch:
`<remap from="/usb_camera/image_raw" to="/camera/image_raw"/>`
no need rebuild nodes