raulmur/ORB_SLAM2 #205: Fail to build ORB-SLAM2

imrasp, 2016-11-26:
Hello, all
I have some trouble during running command "./build.sh". the error is shown below. Thank you for all help in advance.

My working environment is:
OS: Ubuntu 16.04 [mate odroid ux3]
GCC/G++: 5.4.0
Eigen: 3.3~beta1-2

 
> [ 59%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:134: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:110: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:470: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:254: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/MapPoint.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/MapPoint.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:206: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:62: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/System.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:230: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Converter.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Converter.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:350: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 4
> c++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
> CMakeFiles/ORB_SLAM2.dir/build.make:398: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o' failed
> make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o] Error 4
> CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
> make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
> Makefile:83: recipe for target 'all' failed
> make: *** [all] Error 2

cranil, 2016-11-28:
Looks like your compiler is dying so, not sure if this is the same issue I've had. I fixed it by changing [`typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, SparseMatrix::Index> PermutationMatrix;` 
](https://github.com/raulmur/ORB_SLAM2/blob/master/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h#L56) to `typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, SparseMatrix::StorageIndex> PermutationMatrix;` 

- Anil.

callsty, 2016-12-15:
thx cranil :)
Let me add the name of the file in question : 
Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h

typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, SparseMatrix::Index> PermutationMatrix;
to typedef Eigen::PermutationMatrix<Eigen::Dynamic, Eigen::Dynamic, SparseMatrix::StorageIndex> PermutationMatrix;

raktimkashyap, 2018-03-20:
I am having a similar problem..

-- Build files have been written to: /home/raktim/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../Stereo'.  Stop.
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../Mono'.  Stop.
CMakeFiles/Makefile2:718: recipe for target 'CMakeFiles/Mono.dir/all' failed
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../RGBD'.  Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/RGBD.dir/all' failed
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../MonoAR'.  Stop.
CMakeFiles/Makefile2:820: recipe for target 'CMakeFiles/MonoAR.dir/all' failed
make[1]: *** [CMakeFiles/MonoAR.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

selvincephus, 2019-02-19:
The solutions cranil and callsty worked for me. Don't forget to cmake, make and then build after making the changes. :) I forgot, so I am guessing there might be someone who might forget to do that in the end.

shumshersubashgautam, 2020-06-27:
I faced the following error while building ORBSLAM2 plz help me to fix this error.

[ 84%] Linking CXX executable ../Examples/Monocular/mono_kitti
[ 87%] Linking CXX executable ../Examples/Monocular/mono_tum
[ 90%] Linking CXX executable ../Examples/RGB-D/rgbd_tum
[ 93%] Linking CXX executable ../Examples/Stereo/stereo_kitti
[ 96%] Linking CXX executable ../Examples/Stereo/stereo_euroc
[100%] Linking CXX executable ../Examples/Monocular/mono_euroc
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_kitti.dir/build.make:153: recipe for target '../Examples/Monocular/mono_kitti' failed
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
CMakeFiles/Makefile2:109: recipe for target 'CMakeFiles/mono_kitti.dir/all' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_tum.dir/build.make:153: recipe for target '../Examples/Monocular/mono_tum' failed
make[2]: *** [../Examples/Monocular/mono_tum] Error 1
CMakeFiles/Makefile2:146: recipe for target 'CMakeFiles/mono_tum.dir/all' failed
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_tum.dir/build.make:153: recipe for target '../Examples/RGB-D/rgbd_tum' failed
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
CMakeFiles/Makefile2:257: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_kitti.dir/build.make:153: recipe for target '../Examples/Stereo/stereo_kitti' failed
make[2]: *** [../Examples/Stereo/stereo_kitti] Error 1
CMakeFiles/Makefile2:220: recipe for target 'CMakeFiles/stereo_kitti.dir/all' failed
make[1]: *** [CMakeFiles/stereo_kitti.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_euroc.dir/build.make:153: recipe for target '../Examples/Monocular/mono_euroc' failed
make[2]: *** [../Examples/Monocular/mono_euroc] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/mono_euroc.dir/all' failed
make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_euroc.dir/build.make:153: recipe for target '../Examples/Stereo/stereo_euroc' failed
make[2]: *** [../Examples/Stereo/stereo_euroc] Error 1
CMakeFiles/Makefile2:183: recipe for target 'CMakeFiles/stereo_euroc.dir/all' failed
make[1]: *** [CMakeFiles/stereo_euroc.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

windflyxr, 2020-12-28:
> I am having a similar problem..
>
> -- Build files have been written to: /home/raktim/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build
> [ 0%] Built target rospack_genmsg_libexe
> [ 0%] Built target [...]

How did you fix it eventually?

lazy-codebear, 2021-06-04:
I have the same problem.

make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../Stereo'. Stop.
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../Mono'. Stop.
CMakeFiles/Makefile2:718: recipe for target 'CMakeFiles/Mono.dir/all' failed
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../RGBD'. Stop.
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/RGBD.dir/all' failed
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../MonoAR'. Stop.
CMakeFiles/Makefile2:820: recipe for target 'CMakeFiles/MonoAR.dir/all' failed
make[1]: *** [CMakeFiles/MonoAR.dir/all] Error 2
Makefile:127: recipe for target 'all' failed

I could not fix it for a long time.

lazy-codebear, 2021-06-06:
> I have the same problem.
>
> make[2]: *** No rule to make target '../../../../lib/libORB_SLAM2.so', needed by '../Stereo'. Stop.
> CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/Stereo.dir/all' [...]

I fix this bug by run build.sh. After it compiles successfully, the libORB_SLAM2.so will generated in ORB_SLAM2/lib.

I hope it will help you.

MuneebHoda, 2023-11-02:
In file included from /usr/local/include/pangolin/utils/signal_slot.h:3,
                 from /usr/local/include/pangolin/windowing/window.h:35,
                 from /usr/local/include/pangolin/display/display.h:34,
                 from /usr/local/include/pangolin/pangolin.h:38,
                 from /home/muneeb-hoda/testingTelloWorkspace/ORB_SLAM2/include/MapDrawer.h:27,
                 from /home/muneeb-hoda/testingTelloWorkspace/ORB_SLAM2/include/Viewer.h:26,
                 from /home/muneeb-hoda/testingTelloWorkspace/ORB_SLAM2/include/Tracking.h:28,
                 from /home/muneeb-hoda/testingTelloWorkspace/ORB_SLAM2/include/System.h:29,
                 from /home/muneeb-hoda/testingTelloWorkspace/ORB_SLAM2/src/System.cc:23:
/usr/local/include/sigslot/signal.hpp:109:79: error: ‘decay_t’ is not a member of ‘std’; did you mean ‘decay’?
  109 | _compatible_v = detail::is_weak_ptr_compatible<std::decay_t<P>>::value;
      |                                                     ^~~~~~~
      |                                                     decay
/usr/local/include/sigslot/signal.hpp:109:79: error: ‘decay_t’ is not a member of ‘std’; did you mean ‘decay’?
  109 | _compatible_v = detail::is_weak_ptr_compatible<std::decay_t<P>>::value;
      |                                                     ^~~~~~~
      |                                                     decay
/usr/local/include/sigslot/signal.hpp:109:87: error: template argument 1 is invalid
  109 | ble_v = detail::is_weak_ptr_compatible<std::decay_t<P>>::value;
      |                                                     ^

/usr/local/include/sigslot/signal.hpp:109:92: error: ‘::value’ has not been declared
  109 | ble_v = detail::is_weak_ptr_compatible<std::decay_t<P>>::value;
      |                                                          ^~~~~

/usr/local/include/sigslot/signal.hpp:132:53: error: ‘remove_pointer_t’ is not a member of ‘std’; did you mean ‘remove_pointer’?
  132 |                                                std::remove_pointer_t<T>>::value;
      |                                                     ^~~~~~~~~~~~~~~~
      |                                                     remove_pointer
/usr/local/include/sigslot/signal.hpp:132:53: error: ‘remove_pointer_t’ is not a member of ‘std’; did you mean ‘remove_pointer’?
  132 |                                                std::remove_pointer_t<T>>::value;
      |                                                     ^~~~~~~~~~~~~~~~
      |                                                     remove_pointer
/usr/local/include/sigslot/signal.hpp:132:70: error: template argument 2 is invalid
  132 |                               std::remove_pointer_t<T>>::value;
      |                                                     ^

/usr/local/include/sigslot/signal.hpp:132:75: error: ‘::value’ has not been declared
  132 |                               std::remove_pointer_t<T>>::value;
      |                                                          ^~~~~

/usr/local/include/sigslot/signal.hpp:223:32: error: ‘enable_if_t’ is not a member of ‘std’
  223 | struct function_traits<T, std::enable_if_t<trait::is_func_v<T>>> {
      |                                ^~~~~~~~~~~
/usr/local/include/sigslot/signal.hpp:223:32: note: ‘std::enable_if_t’ is only available from C++14 onwards
/usr/local/include/sigslot/signal.hpp:223:32: error: ‘enable_if_t’ is not a member of ‘std’
/usr/local/include/sigslot/signal.hpp:223:32: note: ‘std::enable_if_t’ is only available from C++14 onwards
/usr/local/include/sigslot/signal.hpp:223:62: error: type/value mismatch at argument 2 in template parameter list for ‘template<class T, class> struct sigslot::detail::function_traits’
  223 | uct function_traits<T, std::enable_if_t<trait::is_func_v<T>>> {
      |                                                           ^~

/usr/local/include/sigslot/signal.hpp:223:62: note:   expected a type, got ‘(<expression error> < is_func_v<T>)’
/usr/local/include/sigslot/signal.hpp:223:64: error: expected unqualified-id before ‘>’ token
  223 | uct function_traits<T, std::enable_if_t<trait::is_func_v<T>>> {
      |                                                             ^

/usr/local/include/sigslot/signal.hpp:233:33: error: ‘enable_if_t’ is not a member of ‘std’
  233 | struct function_traits<T*, std::enable_if_t<trait::is_func_v<T>>> {
      |                                 ^~~~~~~~~~~
/usr/local/include/sigslot/signal.hpp:233:33: note: ‘std::enable_if_t’ is only available from C++14 onwards
/usr/local/include/sigslot/signal.hpp:233:33: error: ‘enable_if_t’ is not a member of ‘std’
/usr/local/include/sigslot/signal.hpp:233:33: note: ‘std::enable_if_t’ is only available from C++14 onwards
/usr/local/include/sigslot/signal.hpp:233:63: error: type/value mismatch at argument 2 in template parameter list for ‘template<class T, class> struct sigslot::detail::function_traits’
  233 | ct function_traits<T*, std::enable_if_t<trait::is_func_v<T>>> {
      |                                                           ^~

/usr/local/include/sigslot/signal.hpp:233:63: note:   expected a type, got ‘(<expression error> < is_func_v<T>)’
/usr/local/include/sigslot/signal.hpp:233:65: error: expected unqualified-id before ‘>’ token
  233 | ct function_traits<T*, std::enable_if_t<trait::is_func_v<T>>> {
      |                                                             ^
[... 808 lines cut ...]
[... cut ...]
  167 | EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
      |                                     ^~~~~~~~~~
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/build.make:76: CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:95: CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
muneeb-hoda@muneebhoda-HP-Pavilion-x360-Convert

I am facing a huge list of error, when I try to call the "make" command inside my build of ORB_SLAM2.
Please help me.