raulmur/ORB_SLAM2 #206: Map not showing on ubuntu

doryashar, 2016-11-28:
Hi, i'm running ORB-SLAM2 on my ubuntu 14.04, with monocular version without ROS on the TUM example. the map isn't showing up. please help fixing.
[image]

AlejandroSilvestri, 2016-11-30:
It's odd, it should be running ok.
The map viewer runs on pangolin, and pangolin is an intermediate to opengl.
I would try to update opengl.  I'm guessing.

zhw5772, 2017-07-21:
I met the same problem.
ORB SLAM2, Ubuntu14.04, the map viewer is black.
I have re-installed pangolin and updated opengl, it is still not  working.
Have you solve the problem?
begging for help,thanks....