raulmur/ORB_SLAM2 #207: Problem compiling with new Eigen 3.3 version

dcazzato, 2016-11-29:
Hello, I have been able to compile the code, but since it costed me several hours struggling and reading other issues in order to detect the problem, I report the following solution in order to understand if it depended from an error in my configuration, or if it a possible bug that could be useful for all of you. 

In a few words, I installed the library on a Ubuntu 16.04 LTS, OpenCV 2.4.13 and the latest Eigen 3.3. Well, with this configuration, it is impossible to compile the code, with an error at compile time about the unknown usleep() function. I modified source code by adding #include<unistd.h> in your files LocalMapping, LoopClosing, System, Tracking and Viewer. Although with this trick it compiles, I still found a new error before completing the procedure:

/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY

At this point, the only solution that worked in order to correctly compile the code (even without any needs for modifying source code adding any pre-processor directive as I did before!), is to switch back Eigen library to the version 3.2.10 (stable).

Considering the following open issues:

https://github.com/raulmur/ORB_SLAM2/issues/202
https://github.com/raulmur/ORB_SLAM2/issues/205

I'd hazard there is an incompatibility problem with the new version of the Eigen library.

Hope that my contribution will help (in case I am wrong, please let me know why, in order repeat the installation procedure with latest version of Eigen 3.3 instead the 3.2, also to avoid raising new possible errors at execution time).
 
Regards

hari0920, 2017-01-19:
I faced the same issue and had to use Eigen 3.2.10 as well for a successful compile. Thanks a lot for reporting it!

stihuangyuan, 2017-02-07:
thanks for your share, i have met the same problem, it works.