raulmur/ORB_SLAM2 #209: [ 78%] Linking CXX executable ../Examples/RGB-D/rgbd_tum

chaowu2009, 2016-12-01:
It seems it's almost done. But the last step has issues now. Can somebody help?  Thanks.

It seems  libopencv_core3 is the wrong library build? How should I solve it?
--------------------------------------------------------------------------------------------------------------------
[ 78%] Linking CXX executable ../Examples/RGB-D/rgbd_tum
[ 81%] Linking CXX executable ../Examples/Monocular/mono_euroc
[ 84%] Linking CXX executable ../Examples/Monocular/mono_tum
//usrusr//binbin//ldld::  CMakeFiles/CMakeFilesrgbd_tum.dir//mono_tum.dir/Examples/Monocular/mono_tum.cc.o: undefined Examplesreference to symbol '_ZN2cv6String10deallocateEv'
/opt/ros/kinetic/lib/libopencv_core3.so.3.1: error /adding usrsymbols: DSO missing from command line/
RGB-D/rgbd_tum.cc.o: undefined reference to /symbol '_ZN2cv6String10deallocateEv'
/optbincollect2: error: ld returned 1 exit status
/ros/kinetic/lib/libopencv_core3.so.3.1:/ error addingld symbols:: DSO missing from command line
 CMakeFiles/collect2: error: ld returned 1 exit status
mono_euroc.dir/Examples/Monocular/mono_euroc.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/opt/ros/kinetic/lib/libopencv_core3.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_tum.dir/build.make:135: recipe for target '../Examples/RGB-D/rgbd_tum' failed
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
CMakeFiles/Makefile2:215: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed
CMakeFiles/mono_euroc.dir/build.make:135: recipe for target '../Examples/Monocular/mono_euroc' failed
make[2]: *** [../Examples/Monocular/mono_euroc] Error 1
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mono_euroc.dir/all' failed
make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2
CMakeFiles/mono_tum.dir/build.make:135: recipe for target '../Examples/Monocular/mono_tum' failed
make[2]: *** [../Examples/Monocular/mono_tum] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/mono_tum.dir/all' failed
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
[ 87%] Linking CXX executable ../Examples/Monocular/mono_kitti
/usr/bin/ld: CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/opt/ros/kinetic/lib/libopencv_core3.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/mono_kitti.dir/build.make:135: recipe for target '../Examples/Monocular/mono_kitti' failed
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/mono_kitti.dir/all' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

chaowu2009, 2016-12-01:
Solve it by myself. It is an error of openCV version. 
changed it to the following:
find_package(OpenCV 3.0 REQUIRED)

Ref: https://github.com/raulmur/ORB_SLAM2/issues/13

lolz0r, 2017-01-05:
I had the same issue; your solution solved it. Maybe check in an updated CMakeLists.txt ?