raulmur/ORB_SLAM2 #210: libGLEW missing

chaowu2009, 2016-12-01:
While running the program, got a libGLEW problem while running as following:

cwu@ubuntu:~/project/ORB_SLAM2$ ./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml ~/Downloads/rgbd_dataset_freiburg/
./Examples/Monocular/mono_tum: error while loading shared libraries: libGLEW.so.1.12: cannot open shared object file: No such file or directory

I ran "sudo apt-get install libglew-dev" without any error message.

chaowu2009, 2016-12-02:
ok, solved it myself using the following steps: 
first I used "ldd mono_tum" and find libGLEW is missing because of link error. 
sudo ln -s /usr/local/lib/libGLEW.so /lib/x86_64-linux-gnu/libGLEW.so.1.12

inspirepassion, 2018-09-06:
Maybe your libGLEW.so.1.12 is not in that folder but somewhere else. I encountered similar issue when I try to solve something else (https://github.com/openai/mujoco-py/issues/75#issuecomment-418930523), but in short, I just search where the file is in the system, then either modify the location in ~/.bashrc or create a link as @chaowu2009 said.