raulmur/ORB_SLAM2 #211: g2o optimizer warning while running slam

liming0791, 2016-12-02:
I use the RGB-D interface and the TUM dataset.

While ORB_SLAM is running, I occasionally get some warning message: 

>  virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()

and tracking often lost at the same time, although the program continue to run and can relocate again.

Anyone with the same problem ? Any idea ?

HeatherHu, 2016-12-16:
Did you find where the problem is? I met the same problem in PoseOptimization because after n iterations probably all correspondences is bad. I just add the following check.
        if((nInitialCorrespondences-nBad)<5)
               break;
It fixes my problem. Hope it helps.

AlejandroSilvestri, 2017-03-18:
Hi, I don't know yet the exact cause for this warning, but you should know this is a library warning, g2o warning, non related to orb-slam2, and it seems to be completely harmless.

The library g2o is telling the application orb-slam2 invoked it to optimize something, but provided no vertices to do it, so there's nothing to work on.  It could be avoided with an appropriate check before calling the optimizer, like @HeatherHu 's.

gocarlos, 2017-06-26:
@HeatherHu Could you provide the location where you added this fix?

HeatherHu, 2017-06-27:
@gocarlos I put this in the function Optimizer::PoseOptimization. Try to find the place where it performs 4 optimizations (for loop in there). At the end of each optimization, I add this check to make sure there are still enough vertices in the optimized graph.