raulmur/ORB_SLAM2 #212: Forget to check bad keyframe when running bundle adjustment

randoms, 2016-12-02:
In [line 995](https://github.com/raulmur/ORB_SLAM2/blob/master/src/Optimizer.cc#L995)
Keyframes are added without checking if it is bad. You will get a bad bundle ajustment result if there are bad keyframes. Add the following code will help to fix this issue.
```
if(pKFi == NULL || pKFi->isBad())
    continue;
```

AlejandroSilvestri, 2016-12-07:
You are right!  Thank you.  Hope Raúl add this fix to orb-slam2 code.
I already added it to mine :)

AlejandroSilvestri, 2016-12-30:
In that context, pKF1 is never NULL, and is highly improbably Bad.

pKF1 comes from vpKFs, which contains a vector with a copy content from Map::mspKeyFrames.

Keyframes are erased from mspKeyFrames some lines after setting their bad flag.