raulmur/ORB_SLAM2 #216: Segmentation fault when addEdge in global bundle adjustment

HeatherHu, 2016-12-12:
the error is following:
g2o::HyperGraph::addEdge (this=this@entry=0x7fff77ffdc50, 
    e=e@entry=0x7ffeb8bc40c0) at /usr/include/c++/4.8/bits/stl_tree.h:1803
1803	    find(const _Key& __k) const
(gdb) bt
#0  g2o::HyperGraph::addEdge (this=this@entry=0x7fff77ffdc50, 
    e=e@entry=0x7ffeb8bc40c0) at /usr/include/c++/4.8/bits/stl_tree.h:1803
#1  0x00007ffff52f9787 in g2o::OptimizableGraph::addEdge (
    this=this@entry=0x7fff77ffdc50, e_=e_@entry=0x7ffeb8bc40c0)
    at /home/wendy/catkin_ws/src/orb_slam/g2o/g2o/core/optimizable_graph.cpp:271

Could someone give a hint about the possible reason? Really thanks.

randoms, 2016-12-12:
I have experienced the same issue.  In setVertex, optimizer.vertex(id) can be ```NULL```. Check if optimizer.vertex(id) == NULL before calling setVertex will solve the problem.

before
```
g2o::EdgeStereoSE3ProjectXYZ* e = new g2o::EdgeStereoSE3ProjectXYZ();
e->setVertex(0, dynamic_cast<g2o::OptimizableGraph::Vertex*>(optimizer.vertex(id)));
e->setVertex(1, dynamic_cast<g2o::OptimizableGraph::Vertex*>(optimizer.vertex(pKFi->mnId)));
e->setMeasurement(obs);
optimizer.addEdge(e);
```
after 

```
g2o::EdgeStereoSE3ProjectXYZ* e = new g2o::EdgeStereoSE3ProjectXYZ();
if(optimizer.vertex(id) == NULL || optimizer.vertex(pKFi->mnId) == NULL)
    continue;
e->setVertex(0, dynamic_cast<g2o::OptimizableGraph::Vertex*>(optimizer.vertex(id)));
e->setVertex(1, dynamic_cast<g2o::OptimizableGraph::Vertex*>(optimizer.vertex(pKFi->mnId)));
e->setMeasurement(obs);
optimizer.addEdge(e);
```

HeatherHu, 2016-12-15:
@randoms. Thanks very much. Your answer is really help. It does work for me.

closed 2016-12-15 by HeatherHu

NaderMahmoud, 2018-07-04:
Thanks a lot. Your answer us really helped me!

zhaozhongch, 2021-05-27:
I met this error too after many years lol. Why does this happen?
In my case, a world point can observe keyframe 1,2,3,4,5,6. However, when I add the keyframes vertexes, I only add 2,3,4,5,6. I miss "1". In my code, I add edges according to the point observations. i.e., for an edge, I want to add vertex 1,2,3,4,5,6. Since vertex id ==1 was not added before, so 
```
optimizer.vertex(pKFi->mnId))
```
returns a NULL pointer.

In short, when you meet this problem, check if the vertex with that a certain id has been added to the optimizer before.

dotatw, 2021-07-23:
@randoms  Thanks very much. Your answer is really help. It does work for me.