raulmur/ORB_SLAM2 #218: Tracking failing

OwlCoder, 2016-12-14:
Hi,

I am doing stereo SLAM using ROS with my own stereocameras. The problem I am facing is that my tracker is getting reset again and again. (See below)

Can you guess if this is a software issue or a hardware issue?

Thank You,
Naman

Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 25 points
Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 25 points

ydugua, 2017-10-13:
Hi,

I am using OrbSlam2 stereo but with opencv and a ZED camera.
I facing the same issue. 
Did you find what was the cause of it ?

Thanks a lot

kuangjies, 2018-07-20:
Hi, I meet with the same problem, did anyone solve it? Thanks a lot!

alljiang, 2019-11-03:
Have any of y'all found what was wrong? I'm facing the same issue.
Any help would be great.