raulmur/ORB_SLAM2 #219: Fail to build ROS

callsty, 2016-12-15:
Hi everyone,
I have a problem to compile ROS. When I try this command : 

`cmake .. -DROS_BUILD_TYPE=Release`

it returns : 

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /core/rosbuild/rosbuild.cmake

CMake Error at CMakeLists.txt:4 (rosbuild_init):
  Unknown CMake command "rosbuild_init".

-- Configuring incomplete, errors occurred!
See also "/home/callsty/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".

I don't know where this function is defined. I try grep -R "rosbuild_init" ./ in ORB_SLAM2 but I have found nothing...

Please can you help me?

callsty, 2016-12-21:
Simply, install : 

 ros-indigo-libg2o ros-indigo-cv-bridge

Need to test now.

nloewen, 2017-01-31:
I had a similar issue because my environment wasn't set up correctly.

Try following these instructions: http://wiki.ros.org/kinetic/Installation/Ubuntu#kinetic.2BAC8-Installation.2BAC8-DebEnvironment.Environment_setup

nathangeorge1, 2017-02-16:
In addition to setting up the environment variables correctly, I also had the issue where I was trying to run `sudo ./build_ros.sh`, but doing `sudo` removes the environment variables from what cmake knows. Instead, find the folders that are saying "Permission denied" and run `sudo chown -r yourUserName folderName` on them.

jaminxwong, 2021-06-23:
maybe you could try to run: 
rosdep init && rosdep update
then run: ./build_ros.sh again