raulmur/ORB_SLAM2 #221: Pose estimation using motion model

ank700, 2016-12-17:
Hello, I am using monocular orbslam2. I have seen that, initially 2 frames are used to initialise the system. This transformation is computed by either a homography or fundamental matrix. 
After the creation of the initial map, the motion model is used for pose estimation. The algorithm predicts the location of the feature points in the current image, based on the displacement between the 2 previous frames (mVelocity). If the prediction is true, how is the pose estimated, because now it does not compute For H matrix again?

Thanks

AlejandroSilvestri, 2016-12-21:
mono SLAM compute pose from observations of known 3D points.

Initialization is needed only to provide the first set of 3D points, and never used again.

After that SFM (structure from motion) is applied to compute pose from the observation of known 3D points.  Motion model is used only to narrow the matching area, to gain speed.  If it fails, it retries with a wider matching area, slowing down the system.

Here and there some frames are registered as KeyFrames, stored in memory (frames are ephemeral).  Epipolar geometry is used on pairs of KeyFrames to triangulate new points.

ank700, 2016-12-29:
Thanks a lot @AlejandroSilvestri. 

So, SFM is used to compute pose once the feature matches are found using the motion model. The algorithm used for the pose estimation in SFM is PnP solver?

AlejandroSilvestri, 2016-12-30:
No, ORB-SLAM uses g2o to compute the pose.  It iterates with Levemberg-Marquadt algorithm to minimize the reprojection error, using a Huber kernel to be robust against outliers.

If you wonder about PnPsolver.cc, it uses EPNP but only in relocalisation, not in tracking.