raulmur/ORB_SLAM2 #223: Using mono_tum on custom dataset yeilds no features

kevin-george, 2016-12-22:
I am trying to use the mono_tum.cc AS-IS on my custom monocular dataset as the output is exactly what I want.
This is how I did it,
1. Modified TUM1.yaml with own camera's calibration and distortion coefficients(using OpenCV)
2. Created rgb.txt with correct timestamps and file names
3. Ran mono_tum executable with ORBvoc.txt provided in the library

Is there a step I missed? As the KeyFrameTrajectory.txt is empty and the viewer didn't show any keypoints(green squares) being detected.

This is the stdout when I run the executable,

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 770.723
- fy: 771.437
- cx: 368.152
- cy: 500.103
- k1: 0.207261
- k2: -1.32622
- k3: 2.51224
- p1: -0.0016682
- p2: -0.00137093
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 1321

-------

median tracking time: 0.0582473
mean tracking time: 0.0582658

Saving keyframe trajectory to KeyFrameTrajectory.txt ...

trajectory saved!

shridharkini6, 2018-01-15:
Hi @kevin-george 
Me to getting the same error..
Have you found the solution?
Thanks

kevin-george, 2018-01-16:
Hey @shridharkini6 , I realized that since this library uses Indirect Visual SLAM(ORB feature keypoints), the performance is very dependent on data quality. If the feature detection can't find enough keypoints, the trajectory formation will fail. It's better to test on a live feed and see what works in your environment/scene. Also, this was 2 years ago so I may be missing something :)

closed 2020-02-12 by kevin-george