raulmur/ORB_SLAM2 #224: getting viewing ray vector out of the pose

doryashar, 2016-12-28:
hi,
for a project i'm doing in school i'm using ORB SLAM2.
i have a robot that is driving and mapping. the robot needs to get from it's location to another location with known coordinates, now i have a destination location and self location so i can represent the vector of driving, but i need to know how much i should turn, that is the angle between the robot pose/sight and the driving vector. i have the current pose from the tracker.
how can i find the viewing ray vector / angle i need to turn?
thank you very much for helping

hashten, 2016-12-30:
Hello @doryashar 
You can take the vector in which your goal is by taking the difference between your goal position and your current position. Now you have this vector and your vector of driving, calculate the angle you need to turn by taking the dot product of the normalized vectors:
[image]
Taking arccos of the left-hand expression will give you the angle you need to turn.

doryashar, 2016-12-30:
My question is how to get the vector of driving?

hashten, 2017-01-02:
Good morning @doryashar !
Then I misunderstood your question. So given that you have your pose T_cw specified with a SE(3) transform and your robots forward direction is in its frame's x-direction you could calculate the angle of turning in much the same way as I said in my previous answer. Use the vector towards your target location "vector of driving" and the unit-vector in the direction where the robot is looking, and calculate the angle as in my previous answer. I scrabbled down briefly to illustrate:
[image]

doryashar, 2017-01-02:
First of all, thanks a lot.
as I understand, I just need the first row of the Rotation Matrix (which is
T_cw[1:3,1]), right?
this is assuming that main forward direction is in frame's x direction as
you mentioned, is it always true ?

hashten, 2017-01-02:
You need the whole rotation matrix. It is common to have the camera pointing in the z-direction so the image x- and y-axis is the same as the cameras. This seems to be the case in ORB-SLAM as well!

AlejandroSilvestri, 2017-01-21:
That's right.  Camera points in Z direction, X to the right, and Y points DOWN!