raulmur/ORB_SLAM2 #227: error: Failed to open settings file

test-ing, 2017-01-01:
Hi,

I am trying to run ORB SLAM 2 with ROS on the examples but when I do this:
$ rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml true
I get the following error:
Input sensor was set to: Stereo
Failed to open settings file at: Examples/Stereo/EuRoC.yaml

However when I tried the run ORB SLAM 2 without ROS on the examples, it worked so I guess the issue is not with the settings file.

chaiein, 2017-01-10:
Even I have the same problem.

laxnpander, 2017-01-25:
Doesn't work for me. He prints the correct path as error. Interesting also: I have installed ORB_SLAM2 library as root last year and the version of january 2017 in a folder. With my old library in root it works, with the newer one it doesn't. 
Additionally I have OpenCV 3.1 installed with ros kinetic and OpenCV 2.4 default for other stuff. I think the problem comes from this fact somewhere

test-ing, 2017-01-27:
Try exporting the ros package path and sourcing .bashrc again before running the command

laxnpander, 2017-01-27:
Not solving the problem. Maybe it's relevant, that I link the orb slam library directly not as rospack?

laxnpander, 2017-01-27:
Hrmpf, after reading some issues here I thought the problem can be solved by correct opencv library linking. However I uninstalled my 2.4.9 and now with only 3.1.0 left it still won't work. I even cloned ORB SLAM again, built it with kinetics opencv - still not able to read settings file.

mattgiamou, 2017-02-07:
Hi laxnpander, which version of OpenCV did you compile them with? I believe I am compiling both with OpenCV 3.1.0 but I am still experiencing the problem you've described.

laxnpander, 2017-02-07:
I finally compiled against 3.1.0. Do you have multiple versions of opencv on your computer (ros opencv included)? If you tried to load it with absolute path and are sure that the path is correct, then I the chances are very high it is definitely a mismatching opencv problem. I spent a lot of time on this problem, that's why I am sure :dagger:

mattgiamou, 2017-02-12:
Yes, I still have the ROS Indigo OpenCV version installed, as I need it for other projects. So you were only able to fix it by removing 2.4 from your computer completely, or was there a way in CMake to specify the exact version to load? Thanks for the help!

idolezel, 2017-07-20:
I confirm the solution proposed by laxnpander is correct. Yesterday I got the same issue. Originally I compiled OpenCV 3.1.0 from sources and used it for building the ORB_SLAM2 library. Later I tried to compile main program for ROS and received the Failed to Open settings file issue.

I had to uninstall all OpenCV from my system and install ROS Indigo OpenCV only. Then clean compile the ORB_SLAM2 library and the main program for ROS. Now there are no issues.

lokeshluckybisht, 2018-05-11:
@idolezel & @laxnpander .. thank you for the solution . But I am still unable to get it right. Can you please guide me how to do it rightly. Any help is appreciated. I have removed all the openCV versions and finally re-intsalled ROS with just one openCV . It is still not working.

laxnpander, 2018-05-17:
@lokeshluckybisht : Hmmm, really hard to say. Make sure you pass the absolute path, have the correct OpenCV version and the .yaml is filled with the correct data. I have no information about a bug with FileStorage, therefor it must be somewhere in your general setup. Try more to give us some hints

lokeshluckybisht, 2018-05-18:
@laxnpander thanks. The open CV version was never a problem. I realized it later. I was using a wrong Setting file.

rocker12121, 2022-12-28:
> @laxnpander thanks. The open CV version was never a problem. I realized it later. I was using a wrong Setting file.

How did you set the correct path? I am facing the same problem. Any help is appreciated

1fishingmaster, 2025-03-17:
Input sensor was set to:Monocular [ERROR:0@0.016] global persistence.cpp:531 cv::FileStorage::Impl::open Can not open file:'X？'in read mode Failed to open settings file at:./Example/Monocular/EuRoC.yaml? WHY