raulmur/ORB_SLAM2 #228: Fatal ROS example(with usb camera) Error: close when open usb camera node

xiesiyuan, 2017-01-01:
I have successfully built ORB_SLAM2 under ./build.sh ; meanwhile, I  have successfully build ROS example under ./build_ros.sh. 

Then, I export $ROS_PACKAGE_PATH as PATH/Examples/ROS in ~/.bashrc under root.

After "roscore", I put code as follows:
`rosrun ORB_SLAM2 Mono ~/ORB_SLAM2/Vocabulary/ORBvoc.txt  ~/ORB_SLAM2/Examples/Monocular/TUM1.yaml`
I successfully see the window with the wording "waiting for image"

Then, I use the usb_cam node to call usb camera in ROS:
`rosrun usb_cam usb_cam_node`
To see whether camera is working, I put "rosrun rviz rviz" to call rviz panel and confirm camera is working.

However, fatal error happened!!! after I put in "rosrun usb_cam usb_cam_node", all the ORB_SLAM2 window is closed automatically for no reason. This bug kill all my time today. 

Can anyone help??

xiesiyuan, 2017-01-02:
The following is the log:

```
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
Segmentation fault (core dumped)
```

xiesiyuan, 2017-01-02:
God help me !! Pls!!!

tdnet12434, 2017-01-07:
The monocular node only used for public dataset.
Please take a look at [here](https://github.com/raulmur/ORB_SLAM2#7-ros-examples) instead for work with ROS.

xiesiyuan, 2017-01-09:
@tdnet12434 Are u sure? I saw demo that work for real-time camera.

xiesiyuan, 2017-01-09:
@raulmur Help!~~Help!~~~

jkerfs, 2017-01-17:
I had this same issue, and I resolved it using GDB (in my case I was linking against Opencv4Tegra which appears to have a bug in the FAST feature extractor).

You can compile the source with debugging info by changing the release version to "Debug" in the CMakeLists.txt.  In ROS, you can run the nodes with GDB by following the instructions here: http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Once the segmentation fault occurs, type "backtrace" in the GDB window to see the calls that lead to the fault.  This should help you diagnose the cause of the problem.

hari0920, 2017-01-19:
Seems like you are passing the default settings files. They will have to be changed for your webcam.