raulmur/ORB_SLAM2 #231: Fail to run stereo demo

seusofthd, 2017-01-05:
I met this error when I run the kitti stereo
/usr/local/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

and the calling stack is here:
__Assert_fail
Eigen::internal::plain_array<double, 4, 0, 32>::plain_array DenseStorage.h:128
Eigen::DenseStorage<double, 4, 4, 1, 0>::DenseStorage  DenseStorage.h 187
Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 1, 0, 4, 1>>::PlainObjectBase  PlainObjectBase.h:478
Eigen::Matrix<double, 4, 1, 0, 4, 1>::Matrix  Matrix.h 259
Eigen::Qauternion<double, 0>::Quaternion  Quaternion.h:242
g2o::Sim3::Sim3  sim3.h:53
ORB_SLAM2::LoopClosing::LoopClosing  LoopClosing.cc:38
ORB_SLAM2::System::System  System::94
main

Could anyone help this out?

Thanks,
Di

freedomabcd, 2017-02-19:
Well. I got the same problem, have you fix it up?

seusofthd, 2017-03-05:
It's long time ago and I almost forgot the reason. It should be your dataset problem. You can try monocular dataset first to see whether it works or not

Montura, 2017-03-20:
Look at https://eigen.tuxfamily.org/dox/DenseStorage_8h_source.html,  strings from 61 to 80. 
You can compare your local file /usr/include/eigen3/Eigen/src/Core/DenseStorage.h with that one. Comment the same strings in local file. It help's me to escape this error.