raulmur/ORB_SLAM2 #233: UNITS

aarthi1993, 2017-01-13:
Am sorry for the basic question can anyone juts tell me what are the units for xyz stored in the keyframe trajectory for monocular camera?

AlejandroSilvestri, 2017-01-17:
Arbitrary random unit.
Please see #174

aarthi1993, 2017-01-19:
Thanks AlejandroSilvestri. I also have a doubt is there no way of finding the actual distance traveled  using monocular camera like using some scaling factor?.If so how?

AlejandroSilvestri, 2017-01-19:
There is no way an algorithm can do it, because it hasn't enough data.

On way to provide an scale reference is to indicate the real distance between two 3D points in the map, and then rescale the whole map.  But you must be aware  that monocular SLAM is subject to scale drifting, meaning the scale will change smoothly while you travel away from the reference you provided.

Of course you can use the same proceeding to provide more than one scale reference in different and distant places, but then simply rescaling the map won't work, you'll need to "bundle adjust" it with a specialized bundle adjustment algorithm that can handle such references.

But to begin with an easy procedure, you can travel some known distance (for example 10 m), read distance provided by ORB-SLAM2 and get the scale factor.  You'll need to repeat this each time after map initialization, because each time it picks up a different scale unit.